ouster-lidar/ouster-sdk #1: Request: ability to publish pointcloud2 messages with (epoch-based) Ttimestamps

eric-schleicher, 2018-06-29:
I was able to get ouster OS-1 setup and running quickly.  Also to setup the launch files, start the node and view in RVIZ using fixed frame [os1].  Nice!

I've moved on to integrate with the rest of my ROS system, but running into a small issue with TF and timestamps.  I've rigidly mounted the OS1 and have a static transforms configured (which works as expected), however cannot visualize the PointCloud2s in my TF tree in RVIZ because the pointcloud2 timestamps are "too old"

[image]

Poking through the code, it appears that the ROS node publishes PointCloud2 messages stamp values that are handed over from the driver as sequence (ns) value (based the start of the device or node?), rather than a more traditional ROS epoch-based timestamp.   

It looks like there are two places a change *could* be made, but i thought i would ask here, since  i expect other people will want to visualize ouster data in TF Trees where the time bases are (traditional) epoch based timestamps.  I'm also much more adept at python, and would likely screw it up in c++.

eric-schleicher, 2018-07-21:
For others:  The solution for this is to use ptp4l and properly setup time synchronization on the ROS network.  the user manual covers configuring the unit with plaintext tcp commands. so that it will look for PTP master from which to syncnronize it's time basis.

closed 2018-07-21 by eric-schleicher

oscarmendezm, 2018-08-16:
I'd like to know as well, I've been trying since yesterday with no success.

jepierre, 2018-08-16:
I used the example code provided here: https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/src/os1_ros.cpp and set msg.header.stamp = ros::Time:now() instead of msg.header.stamp.fromNSec(timestamp.count());

oscarmendezm, 2018-08-17:
Hi Jepierre, thanks for contributing to the discussion. I thought about doing that myself, and it would certainly fix things in the short term. However, that makes the message have timestamp of when it was parsed by ROS, not when it was captured. If you are trying to combine this with other sensors and/or odometry it will probably cause issues.

jepierre, 2018-08-17:
@oscarmendezm it was a quick fix for me because in order to create PTP master on my computer, my NIC has to support hw or software timestamping. Which it doesn't. 

I think the steps to @eric-schleicher solutions are: 
* send tcp commands to ouster lidar, if on linux do this:
$ nc <lidar_ip_address> 7501
 set_timestamp_mode TIME_FROM_PTP_1588
 reinitialize
* set up master PTP 
$ ptp4l -i <your NIC> -m

You can check if your NIC support timestamping: 
$ ethtool -T <your NIC>

Unfortunately that didn't work for me since I don't have PTP enabled NIC. you can learn more about PTP here: https://docs.fedoraproject.org/en-US/Fedora/22/html/System_Administrators_Guide/sec-Starting_ptp4l.html

oscarmendezm, 2018-08-22:
Thanks for the instructions jepierr. I tried all of that, but I couldn't get the timestamps to come from the pc. My NIC supports SW timestamps, and as far as I can tell that should be enough?

ZacharyTaylor, 2018-09-12:
Hi all, I just got this running. I modified our fork of the ros driver to always start in ptp mode and reject measurements until a sync has occured https://github.com/ethz-asl/ouster_lidar/pull/1

bochen87, 2018-09-17:
@ZacharyTaylor thanks, that seems to work well!

atleaa, 2019-04-08:
When giving the command "set_timestamp_mode TIME_FROM_PTP_1588" I get "error: Command not recognized."

It may appear that this was removed in v1.11.0?

benripka, 2022-03-14:
Hi there, 
I've run into this issue as well but noticed that I AM able to visualize the point clouds in RViz when I use my base_link, or any other link related to my ouster_sensor link by a static transform, as the fixed frame. It's only when I try to visualize it with the Odom frame that the timing appears to be an issue and block visualization.

I'm new to ROS and was wondering if anyone could explain why this is?

Thanks