ouster-lidar/ouster-sdk #2: Plans to update ouster_ros to follow standard format of ROS messages?

runenordmo, 2018-07-03:
I was just wondering if you have plans to update the ouster_ros example to follow the standard format of ROS messages?

For example, the IMU message format (http://docs.ros.org/melodic/api/sensor_msgs/html/msg/Imu.html), it is stated that accelerations should be in m/s^2 (not in g's). But you appear to have chosen to provide the accelerations in g's?

[image]

Also, the timestamps of the PointCloud2 and Imu messages are currently not given in time since epoch (http://docs.ros.org/melodic/api/std_msgs/html/msg/Header.html).

I would be much appreciated if you could tell me the reasons for the differences from the standard ROS message format, or add an overview over what is different.

dgarbuzov, 2018-07-24:
Good point -- 874b8f7316ffa7e82b05a8b5e6d25eb123764a86 should convert to the proper units.

Re: timestamps, the sensor will only provide absolute time when time syncing is enabled. Otherwise, measurement timestamps will be based on device boot time. You can refer to the OS-1 user manual for how to enable time syncing.

Hope that helps!

runenordmo, 2018-07-24:
Yes, that helps.
Thank you!

closed 2018-07-24 by runenordmo

snakehaihai, 2019-01-14:
What if my hardware does not supprot ptp?  any solution?
 ptp4l -i eno1 -m
interface 'eno1' does not support requested timestamping mode.

Time stamping parameters for eno1:
Capabilities:
	software-transmit     (SOF_TIMESTAMPING_TX_SOFTWARE)
	software-receive      (SOF_TIMESTAMPING_RX_SOFTWARE)
	software-system-clock (SOF_TIMESTAMPING_SOFTWARE)
PTP Hardware Clock: none
Hardware Transmit Timestamp Modes: none
Hardware Receive Filter Modes: none