ouster-lidar/ouster-sdk #10: With VTK-9.0.0 I get this error building the viz_example all other targets build okay

adambarta, 2018-09-03:
MakeFiles/viz.dir/src/viz.cpp.o: In function `vtkRenderingOpenGL_ModuleInit::vtkRenderingOpenGL_ModuleInit()':
viz.cpp:(.text._ZN29vtkRenderingOpenGL_ModuleInitC2Ev[_ZN29vtkRenderingOpenGL_ModuleInitC5Ev]+0xd): undefined reference to `vtkRenderingOpenGL_AutoInit_Construct()'
CMakeFiles/viz.dir/src/viz.cpp.o: In function `vtkRenderingOpenGL_ModuleInit::~vtkRenderingOpenGL_ModuleInit()':
viz.cpp:(.text._ZN29vtkRenderingOpenGL_ModuleInitD2Ev[_ZN29vtkRenderingOpenGL_ModuleInitD5Ev]+0xd): undefined reference to `vtkRenderingOpenGL_AutoInit_Destruct()'
collect2: error: ld returned 1 exit status
CMakeFiles/viz.dir/build.make:249: recipe for target 'viz' failed

adambarta, 2018-09-04:
diff --git a/ouster_viz/CMakeLists.txt b/ouster_viz/CMakeLists.txt
index 5dffccf..cf05a67 100644
--- a/ouster_viz/CMakeLists.txt
+++ b/ouster_viz/CMakeLists.txt
@@ -16,13 +16,15 @@ include_directories(include)
 
 include_directories(SYSTEM
   ${EIGEN3_INCLUDE_DIR}
-  ${VTK_INCLUDE_DIRS}
+  #  ${VTK_INCLUDE_DIRS}
   )
 
+include(${VTK_USE_FILE})
+
 add_executable(viz src/main.cpp src/viz.cpp)
 
 target_link_libraries(viz
-  ouster_client
   ${VTK_LIBRARIES}
+  ouster_client
   ${CMAKE_THREAD_LIBS_INIT}
   )
diff --git a/ouster_viz/src/viz.cpp b/ouster_viz/src/viz.cpp
index 32da72d..778b960 100644
--- a/ouster_viz/src/viz.cpp
+++ b/ouster_viz/src/viz.cpp
@@ -34,7 +34,7 @@
 #include <vtkTransform.h>
 
 #include <vtkAutoInit.h>
-VTK_MODULE_INIT(vtkRenderingOpenGL);
+VTK_MODULE_INIT(vtkRenderingOpenGL2);
 VTK_MODULE_INIT(vtkInteractionStyle)
 
 #include "ouster/colormaps.h"

adambarta, 2018-09-04:
`
diff --git a/ouster_viz/CMakeLists.txt b/ouster_viz/CMakeLists.txt
index 5dffccf..cf05a67 100644
--- a/ouster_viz/CMakeLists.txt
+++ b/ouster_viz/CMakeLists.txt
@@ -16,13 +16,15 @@ include_directories(include)
 
 include_directories(SYSTEM
   ${EIGEN3_INCLUDE_DIR}
-  ${VTK_INCLUDE_DIRS}
+  #  ${VTK_INCLUDE_DIRS}
   )
 
+include(${VTK_USE_FILE})
+
 add_executable(viz src/main.cpp src/viz.cpp)
 
 target_link_libraries(viz
-  ouster_client
   ${VTK_LIBRARIES}
+  ouster_client
   ${CMAKE_THREAD_LIBS_INIT}
   )
diff --git a/ouster_viz/src/viz.cpp b/ouster_viz/src/viz.cpp
index 32da72d..778b960 100644
--- a/ouster_viz/src/viz.cpp
+++ b/ouster_viz/src/viz.cpp
@@ -34,7 +34,7 @@
 #include <vtkTransform.h>
 
 #include <vtkAutoInit.h>
-VTK_MODULE_INIT(vtkRenderingOpenGL);
+VTK_MODULE_INIT(vtkRenderingOpenGL2);
 VTK_MODULE_INIT(vtkInteractionStyle)
 
 #include "ouster/colormaps.h"
`

adambarta, 2018-09-04:
```
diff --git a/ouster_viz/CMakeLists.txt b/ouster_viz/CMakeLists.txt
index 5dffccf..cf05a67 100644
--- a/ouster_viz/CMakeLists.txt
+++ b/ouster_viz/CMakeLists.txt
@@ -16,13 +16,15 @@ include_directories(include)
 
 include_directories(SYSTEM
   ${EIGEN3_INCLUDE_DIR}
-  ${VTK_INCLUDE_DIRS}
+  #  ${VTK_INCLUDE_DIRS}
   )
 
+include(${VTK_USE_FILE})
+
 add_executable(viz src/main.cpp src/viz.cpp)
 
 target_link_libraries(viz
-  ouster_client
   ${VTK_LIBRARIES}
+  ouster_client
   ${CMAKE_THREAD_LIBS_INIT}
   )
diff --git a/ouster_viz/src/viz.cpp b/ouster_viz/src/viz.cpp
index 32da72d..778b960 100644
--- a/ouster_viz/src/viz.cpp
+++ b/ouster_viz/src/viz.cpp
@@ -34,7 +34,7 @@
 #include <vtkTransform.h>
 
 #include <vtkAutoInit.h>
-VTK_MODULE_INIT(vtkRenderingOpenGL);
+VTK_MODULE_INIT(vtkRenderingOpenGL2);
 VTK_MODULE_INIT(vtkInteractionStyle)
 
 #include "ouster/colormaps.h"
```

dgarbuzov, 2018-09-10:
Looks like there's a different opengl module shipped with vtk >= 7.1 We'll update the example to be usable with both, thanks!

closed 2018-09-10 by dgarbuzov