ouster-lidar/ouster-sdk #11: ouster-os1-100sec.bag is 64x1024?

oaix, 2018-09-07:
The [data](https://data.ouster.io/ouster-os1-100sec.bag) you provided is `64 x 1024` for each frame. But your [official website](https://www.ouster.io/product-os1/) shows the resolution is `64 x 2048`.

rafficm, 2018-09-10:
Hi there, the OS-1 can operate at 512, 1024, and 2048 resolution - changing operating modes can be done over TCP. In addition, the sensor can be run at 10 or 20 Hz. We can gather 10x2048 data and post it as well in the next week or so and post it. Customers can change operating modes depending on their situation and use cases and in many cases find x1024 to be sufficient horizontal resolution.

closed 2018-09-10 by rafficm

mrgransky, 2020-01-16:
@rafficm  do you have your `file.bag` and `config.json` opensource?
Which firmware ? (1.13)?