ouster-lidar/ouster-sdk #12: ROS driver IMU message inconsistency - out of sequence msgs and g and m/s² mixed

kajep09, 2018-09-12:
Hello

I have tried to replay these two bag files and looked at the output on the imu topic:

2018-08-29-16-46-17_0.bag
2018-08-29-16-59-48_11.bag

If you echo the imu topic/header/seq you get the following output, where it seems like there are two different contiguous sequences being published.
```
37977
---
8509
---
37978
---
8510
---
37979
---
8511
```

If you look at the full IMU message you can see the same pattern in the seq field, but it also looks like one of the sequences publishes acceleration data in g and the other in m/s².

```
header: 
  seq: 37102
  stamp: 
    secs: 452
    nsecs:  69612610
  frame_id: "os1_imu"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0]
angular_velocity: 
  x: -0.00306263525359
  y: -0.00239684498107
  z: -0.0118510668509
angular_velocity_covariance: [0.0006, 0.0, 0.0, 0.0, 0.0006, 0.0, 0.0, 0.0, 0.0006]
linear_acceleration: 
  x: 0.118408203125
  y: -0.025634765625
  z: -1.01806640625
linear_acceleration_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
---
header: 
  seq: 15005
  stamp: 
    secs: 452
    nsecs:  69612610
  frame_id: "os1_imu"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0, -1.0]
angular_velocity: 
  x: -0.00306263525359
  y: -0.00239684498107
  z: -0.0118510668509
angular_velocity_covariance: [0.0006, 0.0, 0.0, 0.0, 0.0006, 0.0, 0.0, 0.0, 0.0006]
linear_acceleration: 
  x: 1.16118780518
  y: -0.251391174316
  z: -9.98382092285
linear_acceleration_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
---
```

dgarbuzov, 2018-09-13:
Looks like those files contained output recorded from an older version of the example code as well as raw sensor data. We've updated the example data -- does that solve your issue?

kajep09, 2018-09-13:
Thank you for the quick response. The new example data is consistent and fixes the issue.
On a related note, do you have example data from the OS1-16 publicly available anywhere?

kajep09, 2018-09-13:
Instead of creating a new issue/question regarding the IMU messages I will try to ask it here first.

Is it possible for you to provide the proper transformation between the lidar frame and the imu frame? I assume they are co-located one above the other, but I am not sure about the orientation and it would be nice if you could provide an official URDF file or just the proper transforms between the lidar and the imu frames.

closed 2018-09-17 by kajep09

dgarbuzov, 2019-01-29:
The latest update publishes the transformation between the sensor and IMU frames. Hope that helps, and any feedback is appreciated.