ouster-lidar/ouster-sdk #15: [Question] Possibility of setting Ouster IP static

YoshuaNava, 2018-10-23:
Hi,
Is it possible to go over the DHCP setup and just set the Ouster IP static? I have looked at the Software manual, and I can't find any info on this matter.

Thank you.

dsb6063, 2018-10-23:
Hook your OS1 up to a router in dhcp mode. It will assign the address automatically, and should have a name OS1- or OS…

 

Turn off firewalls, and ping the IP address to test for connectivity.         

 

That is the easiest way to get started while you configure the static settings.

 

Thanks,

 

Daryl Banks, PSM

Banks & Banks Consulting, Inc.

Business: (904) 685-2800 <tel:%28904%29%20685-2800> 

Facsimile: (904) 685-2345 <tel:%28904%29%20685-2345> 

Website <http://www.bankssurveyors.com/>   Linked-In <http://www.linkedin.com/pub/daryl-banks/20/5b5/253/> 

 

Banks Family Business 29 years surveying Florida.

dsb6063, 2018-10-24:
> Hey @dsb6063 <https://github.com/dsb6063> ,
> Thank your for your feedback. However, I meant setting the IP static on the Ouster, not allowing it to get an address dynamically. Having a DHCP server always on is an overhead that I would like to skip, if possible.

I can describe a few steps I used. 

 

My initial environment was also challenging to setup for testing. I have a VMWare virtual machine, which required a network bridge configuration to work with the OS1. Having difficulty, I established a private network between two computers (using a local static ip address) verified that connection was working. Then, plugged one network end to the OS1 and powered up. The OS1 is another hardware device and will obtain any IP address assigned to it. Once powered up, you should be able to ping and http://ip address  into the OS1 hardware device.

 

Once you get it working, then you can start the static configuration. I setup on several machines (ubuntu 14,16,and18 and Win 10). There is a dhcp service that I used to assist with one of my setups. 

 

I did use a few software helpers:

 

*	Advanced IP Scanner
*	Wireshark (not so much)
*	DHCP service for windows (helped me bridge one of my connections). It does involve setting up a service.   http://www.dhcpserver.de/cms/download/

 

I’m sure there are better options with the DHCP service. I did get it working, but did eventually uninstall that service. 

 

Thanks,

 

Daryl Banks, PSM

Banks & Banks Consulting, Inc.

Business: (904) 685-2800 <tel:%28904%29%20685-2800> 

Facsimile: (904) 685-2345 <tel:%28904%29%20685-2345> 

Website <http://www.bankssurveyors.com/>   Linked-In <http://www.linkedin.com/pub/daryl-banks/20/5b5/253/> 

 

Banks Family Business 29 years surveying Florida.

dsb6063, 2018-11-10:
Querying the ouster OS1 by TPC by name (os1-991827000128) stopped responding. I had to use the ip address directly. I am adding this information as I have just encountered and forgot the simple solution. 

For example, nc os1-991827000128 7501   did not respond with any commands get_config_txt, get_sensor_info. I had to use nc ip-address 7501.

Dashboard

System Information

boot_id

2154dafc-9b9c-4456-b632-039f4849a81a

ethaddr

BC:0F:A7:00:01:35

fw

ousteros-image-prod-aries-v1.7.0-20180907234232

hostname

os1-991827000128

keeper

v1.7.0

pn

840-101396-03

sn

991827000128

dgarbuzov, 2019-01-29:
Currently the OS-1 only supports network configuration via DHCP.

The easiest way to connect directly to a client machine is to configure your ethernet interface with a static IP (e.g. enp0s31f6 with 192.168.100.1) and run dnsmasq from the command line:

`$ sudo dnsmasq -C /dev/null -kd -F 192.168.100.50,192.168.100.200 -i enp0s31f6 --bind-dynamic`

Hope that helps!

closed 2019-01-29 by dgarbuzov

mrgransky, 2020-01-16:
My system:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.6 LTS
Release:        16.04
Codename:       xenial
```
assign static IP address:
```
Network Connections -->  Wired Connection 1 --> Edit --> IPv4 Settings --> Methods: Manual:
	Address: 10.5.5.1
	Netmask: 255.255.255.0
	GateWay: 10.5.5.1
	Apply --> DONE!!!
```
Sensor works fine!

Problem:
Whenever I turn on the sensor, my internet from WIFI disconnects!
What am I doing wrong?

snakehaihai, 2021-04-27:
> My system:
>
> ```
> $ lsb_release -a
> No LSB modules are available.
> Distributor ID: Ubuntu
> Description:    Ubuntu 16.04.6 LTS
> Release:        16.04
> Codename:       xenial
> ```
>
> assign static IP address: [...]

DO NOT ENTER anything in the gate way. that allows you to use both LIDAR and wifi for global internet.

snakehaihai, 2021-04-27:
NC method does not work for me as well. Any 1 has any solution? 
[image]

snakehaihai, 2021-04-27:
found the solution 

sudo arp-scan --interface=enp59s0 --localnet

will reveal where ouster is hidden at

[image]

This ouster is really good at making people suffer