ouster-lidar/ouster-sdk #22: [Question] Changing the default scanning frequency (lidar_mode)

jelavice, 2018-12-13:
Dear developers,

does anyone know whether it is possible to set the default lidar mode configuration? I can set it using the nc and reinitialize the sensor, after which the frequency is indeed 20 Hz,  however every time I turn on the sensor again it seems to be set to 10 Hz (1024x10).

Has anyone found a way to permanently change this setting?

Thanks,

E

dsb6063, 2018-12-13:
If you want the changes to persist for the next session, you have to write it to the file.

write_config_txt​ will write new values of parameters into a configuration file, so they will take effect after sensor reset. 

reinitialize ​will reinitialize the sensor so the staged values of the parameters will take effect immediately.

write_config_txt

reinitialize

jelavice, 2018-12-13:
Indeed. Somehow missed it in the manual. Thanks for the help.

closed 2018-12-13 by jelavice