ouster-lidar/ouster-sdk #24: os1_node crashes on ARM processor

ilia-sheremet, 2019-01-07:
Hello Ouster team!

I encountered an issues with Eigen library while trying to run **os1_node** on Odroid-UX4 with ARM processor architecture. I compiled and run it previously on x86 without any problems.
Here is the output of the command `roslaunch ouster_ros os1.launch replay:=true` :

dsb6063, 2019-01-07:
I had several issues with the Eigen libraries. You may want to try the newest version of ROS. If that does not work, you may need to rollback the current eigen libs and install the standalone eigen 3 libs. However, this usually leds to other issues to figure out.

dgarbuzov, 2019-01-29:
Building on ARM seemed to work for me -- could you try the latest update and reopen the issue if this is still a problem?

closed 2019-01-29 by dgarbuzov

ilia-sheremet, 2019-01-29:
@dgarbuzov Building was never a problem for me, please check the initial post. 
The problem appears when you run the **OS_1** node. And the problem appears around 4/5 times, so eventually, it works.
Please let me know if you need more logs