ouster-lidar/ouster-sdk #25: publish ROS timestamp with eno device

snakehaihai, 2019-01-14:
I`ve been following this https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/6/html/deployment_guide/s1-using_the_ptp_management_client 
to try sync but two of my PC dont support it 
Any good solution?

snake@snake:~$ ethtool -T eno1
Time stamping parameters for eno1:
Capabilities:
	software-transmit     (SOF_TIMESTAMPING_TX_SOFTWARE)
	software-receive      (SOF_TIMESTAMPING_RX_SOFTWARE)
	software-system-clock (SOF_TIMESTAMPING_SOFTWARE)
PTP Hardware Clock: none
Hardware Transmit Timestamp Modes: none
Hardware Receive Filter Modes: none
snake@snake:~$ ptp4l -i eno1 -m
interface 'eno1' does not support requested timestamping mode.
snake@snake:~$ 

tried other adapter 
snake@snake:~$ ptp4l -i enx00051bd3111d -m
interface 'enx00051bd3111d' does not support requested timestamping mode.
sna

snakehaihai, 2019-01-14:
emm for now. I subscribed to a high-frequency IMU and couple the LIDAR reading to most closest IMU reading. W8ing for your guys to come up with better solution

snakehaihai, 2019-01-15:
Emm I cant. after subscribing to LIDAR and added correct time stamp and then publish. takes too much CPU which cause the CPU to overload. Do you guys have any good suggestion?

dgarbuzov, 2019-01-29:
Besides PTP, you can also use a PPS signal to synchronize with external hardware. Please check your OS-1 manual for time synchronization options.

closed 2019-01-29 by dgarbuzov

mrgransky, 2020-01-16:
@dgarbuzov do we need to open interface box to synchronize GNSS receiver with OS1?
   ```
GNSS          -->       OS1-64
   NMEA_Transmit --> multipurpose_io_mode
   PPS           --> SYNC_PULSE_IN
   PPS_GND       --> GND
```
If so does it terminate the warranty?

snakehaihai, 2020-03-23:
> @dgarbuzov do we need to open interface box to synchronize GNSS receiver with OS1?
> 
> ```
> GNSS          -->       OS1-64
> NMEA_Transmit --> multipurpose_io_mode
> PPS           --> SYNC_PULSE_IN
> PPS_GND       --> GND
> ```
> 
> If so does it terminate the warranty?

its ok. we break all our box :) you have to break it. and cut the cable if possible. the cable too long