ouster-lidar/ouster-sdk #26: ROS Version

jattiogbe, 2019-01-17:
Hello everyone,

I am very new into the world of ROS (catkin, sourcing etc.), I had Ubuntu 18.04 installed recently and ROS-Melodic as well.
Is your system fully compatible with Melodic ?

I am trying to follow and build your examples but I can't make Ros Node work so far. I should be able to launch "`roslaunch ouster_ros os1.launch replay:=true`" but I get this error message:
_

> RLException: [os1.launch] is neither a launch file in package [ouster_ros] nor is [ouster_ros] a launch file name
> The traceback for the exception was written to the log file

_

Can you help me with that? What should I check to see what wreng wrong ?

Thank you.

dsb6063, 2019-01-17:
Attach the log file. The log file will point to where a failure occurred. The last time I reviewed Melodic, it not supported at this time with Ouster. However, many people have successfully installed the driver and are using Ouster with Melodic.

jattiogbe, 2019-01-17:
Thank you for the information, and sorry but I can' find the log file! Where is it supposed to be ? What does its name look like ?

dsb6063, 2019-01-17:
https://answers.ros.org/question/307235/where-are-my-node-log-files/

 

~/.ros/log

jattiogbe, 2019-01-17:
Alright I found it :)
[logs.zip](https://github.com/ouster-lidar/ouster_example/files/2770567/logs.zip)

Here are 4 files I've got in .ros/log.

dsb6063, 2019-01-17:
Your last line shows the node launching. It is a common issue. I can’t remember how I worked around it. 

roslaunch parent running, waiting for process exit. 

Launch again. Once it stops, run roswtf at the commandline. This will run a few tests and show maybe what is the issue with the process.

daniel-lawrence-lu, 2019-01-29:
Have you tried `source ouster_example/ws/devel/setup.bash`? The last time I checked, the code works well with Melodic.

jattiogbe, 2019-01-29:
I finally managed to build the node !
As a beginner, I had made an error with the initialization of the catkin-ws and the various repositories where to clone the sources.
I also needed to keep in mind to source for each new terminal, I'm getting used to it now.

thanks

closed 2019-01-29 by jattiogbe