ouster-lidar/ouster-sdk #29: [error] Cannot start ./viz

Scarabrine, 2019-01-30:
Hi, Ouster,

I just install your lidar on our vehicle's ECU, we run on Ubuntu. After I allocate the IP address correctly and start the visualizer by the command `./viz <os1_hostname> <udp_data_dest_ip>`, there is following error:

./viz os1-991832000882 192.168.1.1
Sensor: os1-991832000882 UDP Destination:192.168.1.1
X Error of failed request: BadValue (integer parameter out of range for operation)
Major opcode of failed request: 155 (GLX)
Minor opcode of failed request: 3 (X_GLXCreateContext)
Value in failed request: 0x0
Serial number of failed request: 39
Current serial number in output stream: 40

I have checked the os1-hostname and ip address are both correct, so what the problem maybe?

More information:
[image]

dgarbuzov, 2019-01-31:
This is an error related to the x server / your graphics drivers. Try running another opengl application like glxgears from mesa-utils; it will fail in the same way.

closed 2019-01-31 by dgarbuzov

Scarabrine, 2019-01-31:
Thank you, I will update my driver.