ouster-lidar/ouster-sdk #30: [Question] Switching to near range lidar mode

salovision, 2019-02-04:
Hi,

Is there a way to switch to enhanced near range mode, perhaps a config command?

Mentioned in specs: "range of 0.0-80m (min range of 0.0m) in enhanced low range mode" and 1.6.0 release notes: "Added new "enhanced near range LIDAR" mode (reduces min range to zero + other enhancements)".

Thanks,
Kim

daniel-lawrence-lu, 2019-02-05:
This can be set by disabling the `window_rejection_enable` setting as described in the user manual. For example you can do it using the [netcat](https://en.wikipedia.org/wiki/Netcat) utility in linux:

    nc os1-991805000142 7501
    window_rejection_enable 0

closed 2019-02-05 by salovision

paxmaniac, 2019-03-09:
The low range performance is pretty horrible. The range readings distort (underestimating distance) when the intensity return is high. I don't know whether there is a way to calibrate to improve this.