ouster-lidar/ouster-sdk #32: Convert .bag to Pcd

messplay, 2019-02-06:
Hi somebody know how can I conver .bag file pcd?  I am trying following 

TErminal 1  roslaunch ouster_ros os1.launch replay:=true viz:=true

Terminal 2 rosbag play --clock <path_to_bag_file> viz:=true

TErminal 3 rosrun pcl_ros bag_to_pcd <path_to_bag_file> ouster_ros/os1_node <path_to_bag_output_folder>

Need I to run roscore before?  Thank you

dsb6063, 2019-02-06:
Here is the conversion. 

 

terminal 1:

roslaunch ouster_ros os1.launch replay:=true

 

terminal 2:

rosbag record /os1_node/points

[ INFO] [1536258906.022794286]: Subscribing to /os1_node/points

[ INFO] [1536258906.025252932]: Recording to 2018-09-06-11-35-06.bag.

 

terminal 3:

rosbag play --clock ~/files/ouster-os1-100sec.bag

 

Then, when recording finishes:

rosrun pcl_ros bag_to_pcd  2018-09-06-11-35-06.bag  os1_node/points ./points

 

 

 

Thanks,

 

Daryl Banks, PSM

Banks & Banks Consulting, Inc.

Business:  <tel:%28904%29%20685-2800> (904) 685-2800

Facsimile:  <tel:%28904%29%20685-2345> (904) 685-2345

 <http://www.bankssurveyors.com/> Website   <http://www.linkedin.com/pub/daryl-banks/20/5b5/253/> Linked-In

 

Banks Family Business 29 years surveying Florida.

messplay, 2019-02-06:
Thank you Daril I am working with a pre-recorded file that I downloaded. I run commad that you send me however folder is empty

dsb6063, 2019-02-06:
Check your file path. Run a  simple test on the file using the same file path. Also, verify it’s executable. 

 

“ls -a ~/Your filepath” 

Chmod +x  (make executable)

messplay, 2019-02-07:
Sorry I can not male the convertion yet, some idea whats happening?. I probed my filepath and chmod nothing happening

messplay, 2019-02-07:
I run rosbag info /home/ticonderoga/catkin_ws/src/ouster_example/ouster_ros/data/2018.bag
path:        /home/ticonderoga/catkin_ws/src/ouster_example/ouster_ros/data/2018.bag
version:     2.0
duration:    1:13s (73s)
start:       Aug 29 2018 18:46:17.85 (1535586377.85)
end:         Aug 29 2018 18:47:31.55 (1535586451.55)
size:        571.0 MB
messages:    54541
compression: none [749/749 chunks]
types:       ouster_ros/PacketMsg [4f7b5949e76f86d01e96b0e33ba9b5e3]
topics:      os1_node/imu_packets      7371 msgs    : ouster_ros/PacketMsg
             os1_node/lidar_packets   47170 msgs    : ouster_ros/PacketMsg

Need I to suscribe to another topics?

messplay, 2019-02-07:
Finally I did it two items to take care
1. node name change to os1_cloud_node/points
2. Re-recorder is obligatory

closed 2019-02-07 by messplay