ouster-lidar/ouster-sdk #33: bag to pcap

messplay, 2019-02-07:
Hi Well yesterday I tried to convert .bag files to pcd file(after a few hours and a lot of help I did it) however my real goal is generate a LAS file from .bag .I supossed start with a pcl file or a pcap would be a good start point. I read in the ouster page that is posible convert to pcap from bag... Any suggestion?

dsb6063, 2019-02-07:
You have to use another software. CloudCompare will have the conversion to LAS.  <https://github.com/notifications/beacon/AY4ZQf6M_rhSkIne3EHtjFwMpukoCyQhks5vLKHmgaJpZM4aqCXy.gif>

jattiogbe, 2019-02-08:
Hi @messplay 
Can you explain (in detail please ; ) )how you got the pcd file using os1's bagfile?
Are you using any SLAM algorithm ?
thanks.

messplay, 2019-02-08:
Hi @jattiogbe Well first of all I am very far from being an expert in ROS. So in advance my apologies.

The first step I took was to identify the topics of the .bag that are on the web page (those that are for download)
rosbag info file.bag
at that point identify that there are basically two topics os1_cloud_node / points and os1_node / points (packet_lidar and imu).

The next thing was to launch the driver
roslaunch ouster_ros os1.launch replay: = true
in another terminal run the downloaded file
rosbag play --clock file.bag
extract / record in a pcd file only the topic that interests me rosbag record os1_cloud_node / points (in another terminal), after it ends there is another .bag file (let's call them file2.bag). if you realize and run the command rosbag info, you see that now only has the topic os1_cloud_node / points and that the message type is pointcloud2, after that you run rosbagplay again
rosbag play --clock file2.bag
and in another terminal
rosrun pcl_ros pointcloud_to_pcd input: = / os1_cloud_node / points

and several pcd files are generated, I think it can be made shorter without giving so many laps but that is achieved yesterday at midnight, now that I write it I think you can extract the file pointcloud2 directly from the file.bag (ie the first file) and run rosrun pcl_ros pointcloud_to_pcd input: = / os1_cloud_node / points directly, but hey, the question is that yesterday I mount it in the cloudcompare those pcd files and they are a complete disaster(but at least already the cloudcompare recognizes them pcd files) (something will be missing (something like geo-referencing maybe hahahaha) or something I did wrong). And if, after this, I run a SLAM algorithm (God, I am very bad in math and worse in maths passed to code) and I will see that it comes out. I hope that I have helped you in something

closed 2019-02-08 by messplay

jattiogbe, 2019-02-11:
Thank you @messplay for the details.
I'll try to go into this deeper in the next few days and let you know.