ouster-lidar/ouster-sdk #35: Communicating with os1

jattiogbe, 2019-02-08:
Hi

I have access to os1 data running os1.launch command, this works fine.
On the other hand, I never could communicate through the netcat commands from the userguide like:
[nc os1-991838000761.local 7501 get_sensor_info](url)

It just keeps running but I get no information in return.
From this info 

> Boost dnsmasq-dhcp[24191]: DHCPACK(enp2s0) 192.168.1.99 bc:0f:a7:00:03:aa os1-991838000761

 , I typed the address instead of os1-... but still nothing.

My firmware is up to date, I can navigate to the sensor hostname and see these informations:

> 
> System Information
> 
> boot_id    314ce424-539a-4b6b-8f73-c237fd0a7478
> ethaddr    BC:0F:A7:00:03:AA
> fw    ousteros-image-prod-aries-v1.10.0-20181211235856
> hostname     os1-991838000761
> keeper    v1.10.0
> pn     840-101396-03
> sn    991838000761

Do you know what I should do ?

tuandle, 2019-02-08:
The command should be : `nc os1-991838000761.local 7501 [enter]` then ` get_sensor_info [enter] `

jattiogbe, 2019-02-08:
alright, it was so simple... it does work better this way.
thank you!

closed 2019-02-08 by jattiogbe