ouster-lidar/ouster-sdk #36: lidar_mode

jattiogbe, 2019-02-11:
Hi,

When I set up the lidar_mode to 1024x20 and reinitialize , I have this confirmation :
```
get_config_param active lidar_mode
1024x20
```
then when I launch `roslaunch ouster_ros os1.launch`
I get these informations that tell me otherwise (1024x10) :

> process[os1_node-1]: started with pid [5646]
> process[os1_cloud_node-2]: started with pid [5647]
> [ INFO] [1549883167.168682915]: Connecting to sensor at os1-991838000761.local...
> [ INFO] [1549883167.169541347]: Sending data to 192.168.1.7 using  lidar_mode: 1024x10
> [ INFO] [1549883167.319664513]: Sensor reconfigured successfully, waiting for data...
> [ INFO] [1549883167.320509210]: Wrote metadata to os1-991838000761.local.json
> [ INFO] [1549883167.321046651]: Sensor sn: 991838000761 firmware rev: v1.10.0

is this normal ?

dsb6063, 2019-02-11:
set_config_param active lidar_mode 1024x20
write_config_txt
reinitialize

## OS1 mode configuration
* Setting the LiDAR horizontal resolution and scan rate
  - Please identify the current sensor IP address or hostname using the steps above
  - The supported values for `lidar_mode` are: 512x10, 1024x10, 2048x10, 512x20, 1024x20. The first value is horizontal resolution and the second is scan rate.
  - Connect to the sensor and reconfigure the `lidar_mode` parameter, storing the modification internally so to use it at next boot. You can skip the `write_config_txt` command to use default value at next boot. `os1-XXXXXXXXXXXX` is the sensor's hostname (`XXXXXXXXXXXX` is replaced by the sensor's serial number).
```bash
  nc os1-XXXXXXXXXXXX 7501
  get_config_param active lidar_mode
  set_config_param lidar_mode 2048x10
  write_config_txt
  reinitialize

tuandle, 2019-02-11:
There is a `lidar_mode` in the launch file. If you don't set it, it will use the default `1024x10` value and overwrite the current mode of the lidar, whether or it you `write_to_txt`. This is defined here 
https://github.com/ouster-lidar/ouster_example/blob/88f08f0ee4f2eb45ded76eb7b733f6eebb4bebce/ouster_client/src/os1.cpp#L295-L297
You might want to go through the code for more details.

jattiogbe, 2019-02-11:
Thanks, 
That's what I did @dsb6063, and I still had 1024x10 when I launched it.
As suggested by @tuandle, I had to set it in the command:
`roslaunch ouster_ros os1.launch os1_hostname:=os1-991838000761.local os1_udp_dest:=192.168.1.7 scan_dur_ns:=50000000 lidar_mode:=1024x20`

closed 2019-02-11 by jattiogbe

mrgransky, 2020-01-16:
@jattiogbe what's `scan_dur_ns:=50000000` ? it does not exist in `os1.launch` !

JimmyFroost, 2020-01-17:
I think it was part of the proper code 1 year ago. It has probaby been modified since then.