ouster-lidar/ouster-sdk #37: Best Practices

dsb6063, 2019-02-11:
What is the best practice to record in the field with the Ouster OS-1?

I have had issues in the past with data loss. Is there an advantage to record to the raw format first, then record to the native ROS compatible topics at the office. 

Issues in the past include packets out of order and loss of packets. 

Thanks,

Daryl

dgarbuzov, 2019-02-11:
For sample data for ROS, we usually just record the packet topic (/os1_node/lidar_packets), which is much lower-bandwidth than the point cloud topic (/os1_cloud_node/points).

There was an issue with the example ROS node not reading packets quickly enough on some hardware in 2048 mode, but that should be fixed. Can you go into more detail about how you're recording data and the data loss you're seeing?

dsb6063, 2019-02-11:
A few months back I used the initial version of the Ouster driver with a laptop with VM Ware installed as the linux host. Running a Cartographer 3D SLAM solution. I was able to generate the pbstream file, but I was having issues with rosbag validate.  I was thinking the VM could be the issue with the network latency. I was recording to the topics os1_node/points and os1_node/imu. In the field, I was experiencing a few 1-3 second gaps in my dataset. It was on a rough dirt road. Maybe vibration caused the gaps in data. 

However, I will record the raw packets from this point forward.  I don’t need real time results.

Old Setup

*	Roof Mount
*	CAT 7 12 ft cable
*	Power Inverter
*	Laptop with VMWare Linux Ubuntu

Next Test

*	Laptop Linux Ubuntu installed no VMWare
*	May switch to a 24 V Battery Source

Any comments would be appreciated. You may save me a few headaches. 

Thanks,

Daryl

dgarbuzov, 2019-03-27:
It's hard to say what else could be causing sporadic missing data like that. You could try capturing network traffic from the sensor directly with e.g. tcpdump to rule out ROS dropping messages. But just vibration shouldn't be causing those kinds of issues -- if you can recreate that behavior and show that the sensor stops sending data, please contact customer service.

closed 2019-03-27 by dgarbuzov