ouster-lidar/ouster-sdk #38: fatal error: ouster_ros/OS1ConfigSrv.h

franzstadt, 2019-02-22:
Hi!
I tried to build ouster_ros according to the readme description, but unfortunately I get the following error:

viz_node.cpp:19:37: fatal error: ouster_ros/OS1ConfigSrv.h: No such file or directory
compilation terminated.
ouster_example/ouster_ros/CMakeFiles/viz_node.dir/build.make:62: recipe for target 'ouster_example/ouster_ros/CMakeFiles/viz_node.dir/src/viz_node.cpp.o' failed
make[2]: *** [ouster_example/ouster_ros/CMakeFiles/viz_node.dir/src/viz_node.cpp.o] Error 1
CMakeFiles/Makefile2:1570: recipe for target 'ouster_example/ouster_ros/CMakeFiles/viz_node.dir/all' failed
make[1]: *** [ouster_example/ouster_ros/CMakeFiles/viz_node.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1 -l1" failed

Can you help me?

Thanks forward!

Kind regards, Gabor

vitosansa, 2019-03-06:
Hello,
I've got the same problem..
Can you help me?
Thanks in advance

dsb6063, 2019-03-07:
Have you tried installing the ouster driver on another platform or after a clean install?

paxmaniac, 2019-03-09:
Same issue here: clean install.

paxmaniac, 2019-03-09:
I reverted back to 9a7ffc5644956bf4846488a3dba3650827ffbe0b and was able to build.

daniel-lawrence-lu, 2019-03-09:
Try adding

```
add_dependencies(os1_node ${PROJECT_NAME}_gencpp)
add_dependencies(os1_cloud_node ${PROJECT_NAME}_gencpp)
add_dependencies(viz_node ${PROJECT_NAME}_gencpp)
add_dependencies(img_node ${PROJECT_NAME}_gencpp)
```

to the bottom of [`ouster_ros/CMakeLists.txt`](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/CMakeLists.txt)

dgarbuzov, 2019-03-27:
Should be fixed in  4ce8e4f

closed 2019-03-27 by dgarbuzov