ouster-lidar/ouster-sdk #40: Timing offset of points

zarmomin, 2019-02-27:
Since the update, the timing offsets are incorrect. One entire revolution (@10Hz) is assigned the same offset, and the behavior near x=0 is completely off the charts.

[image]

I believe some of the issues arise from an invalid int64 to float conversion of (ts - scan_ts), but that still does not explain the behavior in the x=0 area:

https://github.com/ouster-lidar/ouster_example/blob/88f08f0ee4f2eb45ded76eb7b733f6eebb4bebce/ouster_client/include/ouster/os1_util.h#L117-L127

andre-nguyen, 2019-03-05:
You should contact their support email directly about this. I'm curious if this affects all users.

inkyusa, 2019-03-08:
@zarmomin I am just curious what is "the update" you referred to? (e.g., Could you point out any commit)?

zarmomin, 2019-03-08:
Sure. It's commit 88f08f0ee4f2eb45ded76eb7b733f6eebb4bebce, specifically this is how the timing was computed before (line 95):
https://github.com/ouster-lidar/ouster_example/commit/88f08f0ee4f2eb45ded76eb7b733f6eebb4bebce#diff-08cf0769b0ecd852e95ec27e04233154L90

dgarbuzov, 2019-03-13:
As far as I know, the conversion is legal.  The output you're seeing is just rviz guessing the range of the data incorrectly, and mapping the palette to the first few values.  If you uncheck "autocompute intensity bounds" and set the range to [0, 1e8], you should see more reasonable output.

The meaning of the timestamp field on points did, however, inadvertently change in the commit you pointed out. It used to be a fraction of the "scan duration", but turned into nanoseconds represented as a float. I think the former is often used for point cloud data batched by time, but might not make sense in the context of this example code.

It would be simplest to just use an integer timestamp of nanoseconds after the start of the scan (and it would avoid any loss of precision), but it might be surprising to users expecting a fractional timestamp. Do you have a preference?

dgarbuzov, 2019-03-27:
Switched to integer timestamps; please reopen if you have any suggestions

closed 2019-03-27 by dgarbuzov