ouster-lidar/ouster-sdk #43: Connecting Issue with New Ouster os1-xxxxxxxxxx

HappySamuel, 2019-03-08:
Hi,

I just received the ouster 3D-lidar os1-16A. I connected the lidar to the Industrial PC, which come along with built-in 6 ethernet ports. I plugged the os1 into the IPC, but i couldn't run the os1 with ROS. I read the software manual, but still failed to get it hooked up.

Inside the software manual, it's written:
The sensor is configured to dynamically obtain an IP address using a DHCP server. If the
sensor is plugged into a network switch, it will automatically obtain an IP and you can find that
IP by checking the DNS leases on the network switch

Network setup:
ethernet is set to manual ip address: 192.168.1.1

ROS Launch file setup:
os1_hostname: os1-xxxxxxxxxxxx.local
os1_udp_dest: 192.168.1.1

but i couldn't find the ip_address of the sensors, therefore i try use the hostname, instead of using ip-address.

Besides, I tried to enter the following command into terminal:
nc os1-xxxxxxxxxxxx.local 7501
but i get nothing in return.

Therefore, I need some guides on how to hook it up or how to identify the sensor ip-address.

Samuel

paxmaniac, 2019-03-09:
Did you install a DNS server on your machine (dnsmasq)? The instructions for this are in thee software manual.

HappySamuel, 2019-03-20:
Hi, @paxmaniac 

Sorry for the late reply. I finally addressed the connection problem between ouster OS-1 and either direct to PC or through network switch/router. It's done according what you mentioned by following the software manual carefully. Thank you.

Samuel

closed 2019-03-20 by HappySamuel

dkeeton2018, 2019-05-18:
@HappySamuel I'm coming across a similar issue and have followed the guide instructions with no luck. Did you ever have a failure to connect with the client IP?
Failed to connect to client at: 192.168.0.1
I keep getting this error and curious if you came across it and how you solved. Thank you.

HappySamuel, 2019-05-21:
Hi @dkeeton2018 

First, is your ouster lidar connected directly to your pc or via a router/switch  ??
Second, What is the ip-address of your pc, and ip-address of your lidar  ??
Third, has you tried with the command: nc os1-xxxxxxxxxxxx 7501   ??

Best Regards,
Samuel

dkeeton2018, 2019-05-22:
Thanks, I was able to solve on Sunday.

HappySamuel, 2019-06-12:
Hi @antithing 

If you received your lidar recently, which mean it comes along with firmware 1.12. Then it should be easy.

First, you can set your ethernet connection to become manual and assign ip-address and netmask, and gateway too. Example as below picture:

[image]

Second, you open your os1.launch and edit the os1_hostname to be os1-xxxxxxxx.local, together with os1_udp_dest to be the ip-address you set for the manual ethernet connection.

[image]

Then, it shall be done by now. save and try to run the launch file again.

Best,
Samuel

antithing, 2019-06-12:
Thank you! I am on windows, without ros, I can ping the os1 using the serial, but i need the actual IP. 

is there a way to get the IP somehow?

thanks!

HappySamuel, 2019-06-12:
Hi @antithing 

Since you are using it on windows, i couldn't help. Maybe you need to refer to the software manual. Or send email to Ouster and ask help from their technician.

Best,
Samuel

JimmyFroost, 2019-12-10:
Hello @antithing,

I am trying to make it work on Windows as well.
Have you been successfull since last June ?

antithing, 2019-12-10:
Hi @JimmyFroost, no unfortunately I had to give up on this hardware, as we were unable to get the data streaming in windows. Good luck! If you do solve it, please post it here!

dsb6063, 2019-12-10:
Hi @JimmyFroost <https://github.com/JimmyFroost> ,

 

You may want to consider outsourcing. If you are an individual, try a kickstarter if there is enough interest in this particular problem. The developers at Kitware have already solved this issue. Kitware developed the Ouster Studio Software. I say a kickstarter since many developers on the forum have the same issues who would like to contribute. With a few kickstarter projects generating enough income to fund these issues. I have a business but little time. I have to use outside help for these tougher issues.  

 

Don’t give up. Just change the path. Good Luck!

JimmyFroost, 2019-12-12:
So far, I can only view a "map" of lidar data through OusterStudio on windows :

[image]

Then, even if I have a target ip for lidar and IMU data, I am struggling with DNS mask, DHCP server, which I don't get at all...

[image]