ouster-lidar/ouster-sdk #44: Any Gazebo URDF Model provided?

HappySamuel, 2019-03-13:
Hi Ouster,

Is there any URDF model provided for gazebo simulation???

Best Regards,
Samuel

dgarbuzov, 2019-03-27:
Unfortunately no, though we welcome contributions!

closed 2019-03-27 by dgarbuzov

wilselby, 2019-03-27:
Try [this](https://github.com/wilselby/ouster_example/blob/master/ouster_description/urdf/OS1-64.urdf.xacro):

HappySamuel, 2019-03-27:
Hi @wilselby 

Thank you very much for it. I will test it out soon.

Best,
Samuel

HappySamuel, 2019-04-02:
Hi @wilselby 

I tried but i can only see your ouster os-1-64 model in RVIZ, there's no model in Gazebo. Do you know how i can visualize its model in Gazebo too?

Best,
Samuel

wilselby, 2019-04-03:
I had it working on ROS kinetic and the associated Gazebo version but it may have issues on more recent versions of Gazebo (though I'm not sure why).  Let me know if you find out and I'll post here if I get around to integrating with Gazebo 9.

HappySamuel, 2019-04-03:
Hi @wilselby 

I am using the Gazebo 9. I will update you too if i found a solution on this.

Best,
Samuel

wilselby, 2019-05-11:
@HappySamuel I uploaded a new .dae file that works in Gazebo 9.

HappySamuel, 2020-09-16:
Hi @wilselby 

Thanks for the update. It looks really nice.

Best,
Samuel