ouster-lidar/ouster-sdk #45: Issue reading pointCloud in Matlab

21Ansh, 2019-03-21:
I am using rosrun pcl_ros pointcloud_to_pcd input:=/ouster/points for saving point cloud to pcd files coming from rosbag. When I tried to read these pcd files in matlab using pcread I got following output

[image]

[image]

daniel-lawrence-lu, 2019-03-21:
Try reshaping the images to `[1024, 64]` instead of `[64, 1024]`.

21Ansh, 2019-03-21:
tried, not working and ideally, it should be 64,1024 because vertical resolution of ouster lidar is 64.

dgarbuzov, 2019-03-27:
The issue is that the point clouds follow the staggered firing pattern of the sensor and are not arranged in columns according to the azimuth angle as you expect.

You can infer how this works from the angle offsets described in the user guide, but it might be easier to just look at the sample code that produces ROS image message output:

https://github.com/ouster-lidar/ouster_example/blob/4ce8e4ff7e4960bf6bc356e0c590ce2df404835f/ouster_ros/src/img_node.cpp#L88-L103

Hope that helps!

jattiogbe, 2019-04-11:
Hello @21Ansh ,

Did you solve your case using Matlab ?
I have an OS-1.16, and I am a beginner in Matlab, so I'm thinking about the use of both together.

Thank you.

closed 2019-05-01 by dgarbuzov