ouster-lidar/ouster-sdk #46: OS1 ROS nodes as nodelets

runenordmo, 2019-03-24:
Have you considered also providing a nodelet implementation of the `os1_node` and the `os1_cloud_node`?

wsascha, 2019-04-12:
You could also "just" create a nodelet and start it as separate process, e.g. using the "standalone" tag in the launch xml.

runenordmo, 2019-04-12:
I would probably start the implemented nodelets with a nodelet manager from the launch file

wsascha, 2019-04-12:
Yeah, just saying that the guys from Ouster can also just implement only a nodelet and we get both nodelets in nodelet managers and standalone nodelets. If I find time in the next week I'll create a PR

runenordmo, 2019-04-12:
Ok, I see. Would be great if you could create a PR on it :+1:

dgarbuzov, 2019-05-01:
Good to know that's something people are interested in, though we don't have any plans to switch the sample code to nodelets. Would be happy to consider a PR as long as it doesn't make the code significantly harder to understand.

Are you mostly concerned with the serialization overhead? Is the performance of the sample code problematic? Would nodelets make it easier to integrate with existing projects?

runenordmo, 2019-05-02:
Thank you for your answer, @dgarbuzov . Yes, I am mainly concerned with the copy cost when sending the messages. I think it could improve the performance of the sample code. The sample code is quite easy to integrate with existing projects. However, this would make it easier to integrate in existing projects based on nodelets / nodelet managers.

mohit-khosti, 2019-07-29:
Hi @dgarbuzov 
You have a great knowledge about Ouster LiDAR sensors, so it would be very nice of you to join our Ouster forum https://forum.ouster.at/ where you can contribute to this newly created support facility for Ouster LiDAR!

erlenner, 2019-08-09:
Seems like these guys are working on this feature: https://github.com/ethz-asl/ouster_lidar/commits/feature/nodelets

daniel-dsouza, 2019-08-28:
you can also try my fork of the ouster driver: https://github.com/swri-robotics/os1_driver

closed 2020-03-11 by dgarbuzov

msadowski, 2020-07-11:
I know this issue is closed but I would second developing a nodelet support. This should add quite a bit of value for anyone looking into filtering laser scan data because of the zero copy mechanism.

With ROS you can support both nodes and nodelets at the same time. One project that I know of that did it is VESC driver by mit-racecar team: https://github.com/mit-racecar/vesc/tree/master/vesc_driver