ouster-lidar/ouster-sdk #47: neither a launchfile...

benIflock, 2019-04-07:
more of an issue with the instructions for the ROS build - following it to the letter and get this error when i try and run 

[os1.launch] is neither a launch file in package [ouster_ros] nor is [ouster_ros] a launch file

daniel-lawrence-lu, 2019-04-08:
Did you follow the first step in the ouster ros readme file?

> Make sure to set up the ROS environment with `source /path/to/myworkspace/devel/setup.bash` in a new terminal

benIflock, 2019-04-08:
Hello Daniel

Do You mean after I have built the workspace but before I then launch the
os1  - yeah I have been trying these steps for a full day now and getting
nothing - I suspect it might be something around the actual build command -
I am working from the downloaded file structure from the git and have tried
pointing to build to both the ouster_ros dir and the ouster_example dir
(which is what the documentation suggests) neither seem to generate launch
file however - just the three folders in the workspace

I have since managed to build a seemingly working example based on what
feels like a fork of your examples as the launch commands are the same And
it opens rviz but I don’t seem to see anything and I feels like I am not
getting any data in - when I try and run the get_config_param udp_ip it
returns null even if I set it via ncat and write it -
Does any of this offer any clues!
Been at this a couple of days now
Thanks
B

dgarbuzov, 2019-05-01:
Think this was just an issue symlinking the repo into the workspace.

closed 2019-05-01 by dgarbuzov

carballodaniel, 2019-06-18:
I had the same issue

Fixed by switching from "...ln -s..." to "...ln -sr..."