ouster-lidar/ouster-sdk #48: OS1 Timestamp Inconsistent for PointClouds and TF

wsolichin-bto, 2019-04-16:
It seems there is inconsistency in the timestamps for the TF in LIDAR and the point clouds, and it causes TF to be used appropriately. In [line 67 here](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/src/os1_ros.cpp#L67), the time is obtained from the os1_cloud_node.cpp, which is defined as `std::chrono::nanoseconds{scan_ts}` (Using Chrono for Time), but the TF, as specified in [line 83 here](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/src/os1_ros.cpp#L83) is using ROS Time, and therefore porting it to a SLAM algorithm such as [HDL Graph SLAM](https://github.com/koide3/hdl_graph_slam) with a ROS bag will cause TF error. Should the Ouster LIDAR ROS follows the ROS Time consistency for it to be used in ROS?

Thank you for your help.

tuandle, 2019-04-17:
Not sure if I understand the problem correctly but to me the TF you mention (you mean the transform from frame "os1_sensor" to "os1_lidar" and "imu" right?) is _static_. So I guess the timestamp is less important here because you can just obtain it once and make it constant instead of subscribing to the topic.

wsolichin-bto, 2019-04-17:
Sorry my issue wasn't clear, I meant that when an external node subscribes to this OS-1 point cloud topic, it is not possible to get the TF displayed in RViz as the point cloud topics is not based on Epoch time stamps. The solution was solved when using the PTP with the sensors connected to the hardware as specified in this https://github.com/ouster-lidar/ouster_example/issues/1 , but when playing ROS Bags recording the packets, then the TF issue still arises like mentioned there. Do you have any suggestions on what to be done for ROS bags to be played correctly with the TF displayed in RViz?

tuandle, 2019-04-21:
Sorry for the late reply. If you work with rosbag and the bag contains only os1_lidar_packages, i.e NOT the raw point cloud, you can change the driver a bit when processing those packages and output the time stamp of ros.  You should change these lines: [L#33](https://github.com/ouster-lidar/ouster_example/blob/4ce8e4ff7e4960bf6bc356e0c590ce2df404835f/ouster_ros/src/os1_ros.cpp#L33), [L#67](https://github.com/ouster-lidar/ouster_example/blob/4ce8e4ff7e4960bf6bc356e0c590ce2df404835f/ouster_ros/src/os1_ros.cpp#L67) to use `ros::Time::now()`.
Remember to revert the changes when you work with the real Ouster.
Hope this helps!

wsolichin-bto, 2019-04-23:
Thanks for the suggestions for working with the ROS Bags. I changed those lines as you suggested and it definitely helps!

closed 2019-04-23 by wsolichin-bto

HappySamuel, 2019-06-18:
Hi @tuandle 

If i want to have the raw point cloud, to use ros::Time::now(), what shall i do? Because i am facing the similar problem as @wsolichin-bto 

Best,
Samuel

tpet, 2019-09-20:
The problem seems to be that when the sample data were recorded, the lidar was not synchronized with the system/ROS clock. Would it be possible to record new sample data with lidar synchronized properly? I guess in that case the timestamps in the lidar packets would match the rest of the system (and of the bag file) and one could replay, process ,and visualize all data propertly, is it true?