ouster-lidar/ouster-sdk #49: how to convert the .bag into pcd or ply

XiaoshuiHuang, 2019-04-17:
I plan to use the sensor to capture data for point cloud algorithm. So, I need to convert the *.bag files into pcd or ply files. Does anyone know how to convert the bag files from ouster sensor into pcd or ply files?

runenordmo, 2019-04-18:
`pcl_ros` has a ROS node called [bag_to_pcd](http://wiki.ros.org/pcl_ros#bag_to_pcd)

dsb6063, 2019-04-18:
Also,

 

It should already be posted. You may search the previous posts if there are issues with the below commands. Here is the conversion. 

 

terminal 1:

roslaunch ouster_ros os1.launch replay:=true

 

terminal 2:

rosbag record /os1_node/points

[ INFO] [1536258906.022794286]: Subscribing to /os1_node/points

[ INFO] [1536258906.025252932]: Recording to 2018-09-06-11-35-06.bag.

 

terminal 3:

rosbag play --clock ~/files/ouster-os1-100sec.bag

 

Then, when recording finishes:

rosrun pcl_ros bag_to_pcd  2018-09-06-11-35-06.bag  os1_node/points ./points

 

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

XiaoshuiHuang, 2019-04-24:
OK.  I tried the solution. The problem is solved. Thanks.

closed 2019-05-01 by dgarbuzov

cgb09c, 2020-06-04:
Just a correction, the name of the topic to subscribe to is /os1_cloud_node/points not /os1_node/points

rosbag record /os1_cloud_node/points