ouster-lidar/ouster-sdk #51: OS1-16 channel Lidar appearing like a 64 channel?

purplefishies, 2019-04-18:
Hi,

We're having a problem where our Os1-16 channel Lidar is creating point clouds that resemble what a 64 channel Lidar would produce.  Specifically , the parameter H appears to be hard coded and is not something that is determined by querying the service ( as Width is determined in os1_cloud_node.cpp ).

We are getting much more data that we should be expecting.

dgarbuzov, 2019-05-01:
The OS1-16 outputs udp data in the same format, but only 16 beams will have real data (the rest should be zeroed out). This adds some network overhead but simplifies the client code.

closed 2019-05-01 by dgarbuzov

HappySamuel, 2020-03-27:
Hi @purplefishies 

Have you sort out how to get 16 channel lidar points instead of 64 channels?
How shall i implement this filtering in the ouster code? Right before it publish out the cloud as ROS topic, which will eventually create big bag files when do the rosbag recording.

Best,
Samuel

bhomaidan1990, 2023-05-24:
@HappySamuel I think that you can do it like this:

```cpp
OUT_CHANNELS = 16
int DIV_ = 64 / OUT_CHANNELS

void cloud_filter(pcl::PointCloud<OusterPoint>::Ptr cloud){
    if(cloud->height > OUT_CHANNELS){
    pcl::PointIndices::Ptr inliers(new pcl::PointIndices());
    pcl::ExtractIndices<OusterPoint> extract;
    for(unsigned int i = 0; i < (*cloud).size(); i++){
        if ((static_cast<int>(cloud->points[i].ring) % DIV_ != 0)){
            inliers->indices.push_back(i);
        }
    }
    extract.setInputCloud(cloud);
    extract.setIndices(inliers);
    extract.setNegative(true);
    extract.filter(*cloud);
}
}
```