ouster-lidar/ouster-sdk #55: os1_node.cpp crashing due to JSON error

tjblada, 2019-05-16:
After updating Ubuntu 18.04, os1_node.cpp now gives a segfault when running the lidar code. Here is my stacktrace:

Starting program: /home/USER/rosdev/catkin_ws/devel/lib/ouster_ros/os1_node
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff71bb350 in ?? () from /usr/lib/x86_64-linux-gnu/libjsoncpp.so.1

tjblada, 2019-05-16:
Spent a bit of time narrowing it down to Json::CharReaderBuilder builder{}; line in os1.cpp.

dllu, 2019-05-17:
You may need to clean the build and recompile the project whenever you update your operating system.

[Ubuntu updates may cause ABI breaks](https://askubuntu.com/questions/249425/what-guarantees-does-ubuntu-provide-concerning-binary-compatibility-across-relea).

closed 2020-03-11 by dgarbuzov