ouster-lidar/ouster-sdk #59: Corner cases

asherikov, 2019-05-24:
It appears that in all three of the following cases the lidar readings are zeroed out:
1. an obstacle is too close
2. an obstacle is out of range
3. azimuth block is bad
Am I correct? Is it possible to discreminate these cases somehow?

dgarbuzov, 2020-03-11:
Currently there's no way to distinguish a lack of a return from a return being within the minimum range.

A bad azimuth block can be identified using the "azimuth data block status" field in each azimuth block. See the "lidar data format" section in your user guide.

closed 2020-03-11 by dgarbuzov

asherikov, 2020-03-11:
Block status info is missing in the point cloud, so basically all three cases are represented identically when using point clouds.

dgarbuzov, 2020-03-11:
Correct me if I'm wrong, but I think `batch_to_iter` will skip adding points for invalid azimuth blocks entirely, whereas the other cases will show up as points at the origin.

Not sure that's super useful -- what would you like to see in the point cloud output?

asherikov, 2020-03-11:
Probably, I haven't worked with it for a while. 
I've already refactored the node to match my specific needs. In general, I think, some sort of a range image could be more useful than a point cloud.