ouster-lidar/ouster-sdk #60: Conversion of bag files to las files

Ashik19, 2019-05-26:
I have made my rosbag by recording the /os1_cloud_node/points. However, when I am converting them to pcd file using the 
rosrun pcl_ros bag_to_pcd s.bag /os1_cloud_node/points ./home command. It’s converting the bag to pcd but all the pcd file has the same size of 1.6mb and also no matter the duration of my recording every time it’s saying it has got 32768 points data to convert into pcd file. Can anyone please tell me how I can record the points topic and convert them into las file ?

TaylorDixonHD, 2019-05-26:
If your end goal is singular sweeps to las format i would advise writing your own node that subscribes to /os1_cloud_node/points and then utilizes libLAS to format and write the las files within the subscriber callback. Then run the newly created node within the environment at the same time running the bag file.

Ashik19, 2019-05-26:
Thank you very much for your kind reply. But can you please tell me a little bit detail ? Correct me if I am wrong do I will create a new node subscribing to the os1_cloud_node/point topics utilising the liblas format. And this will record the file in the las format ?

Ashik19, 2019-05-26:
I was thinking to create a bag file then convert it to pcd and later to las by using the pdal. But I am facing the problem when I am converting it to pcd all the pcd files are getting same size of (1.6mb).

closed 2020-03-11 by dgarbuzov