ouster-lidar/ouster-sdk #62: Bag to Pcap

Ashik19, 2019-06-03:
Hi,
I have recorded a bag file with the (pointcloud2 topic) os1_cloud_node/points in the herewith attached bag file. 

https://drive.google.com/file/d/1Fn2VSB96u5LI1FmaRQJbYn_udV7TkjBZ/view?usp=sharing

Can anyone please tell me how I can convert this bag file into pcap files.I am using ouster os_16 LiDAR unit?

Thanks
Ashik

peyman-sabouri, 2019-08-08:
@ghost I have same question. Did you find the answer?

mrgransky, 2020-01-15:
OusterStudio can record original `.pcap` file if that might help your application. Then a reverse application (`.pcap --> .bag`) [here](https://answers.ros.org/question/268403/convert-pcap-to-bag-file/) can be performed.

But you may wanna try [this](https://github.com/velodynevls/velodyne_vls/tree/master/bag_to_pcap)  for (`.bag --> .pcap`).

Please note that both of the aforementioned links are suggested for velodyne. Therefore, modifications are required.

dgarbuzov, 2020-03-11:
Unfortunately, we don't currently provide sample code to do this

closed 2020-03-11 by dgarbuzov