ouster-lidar/ouster-sdk #67: Will there be any incompatibility issue between Ouster Firmware v1.10 and Ouster-ros v1.12?

HappySamuel, 2019-06-13:
Hi,

I just faced a issue, that when i start the latest ouster-ros (v1.12) with ouster OS1-16 lidar firmware (v1.10) and found  the following behavior:

[image]

As you can see, there's no reading on the environment distance at all. There's nothing when i turn on the rviz, i couldn't see any pointcloud distribution. The ouster imu is working just fine.

Please advice what shall i do. Thanks.

Best,
Samuel

tuandle, 2019-06-13:
Have you tried to update your firmware?

HappySamuel, 2019-06-13:
Hi @tuandle 

Not yet. Will firmware upgrading solve this problem? If it will, then that means there's incompatibility issue between ouster-ros v1.12 package and ouster lidar firmware v1.10.

Best,
Samuel

tuandle, 2019-06-13:
Usually, you should use a driver version with **compatible** firmware. If you read the [changelog](https://github.com/ouster-lidar/ouster_example/blob/master/CHANGELOG.md) you will see what has been changed in the firmware AND the driver to avoid compatibility issue

HappySamuel, 2019-06-14:
Hi @tuandle 

I have test it in 2 ways according to your suggestion.
(1) use ouster-ros (v1.10) with the ouster lidar firmware (v1.10):
Result: Problem still exist as picture as in the first thread.
(2) upgrade ouster lidar firmware to v1.12 and use it with ouster-ros (v1.12):
Result: Problem still exist as picture as in the first thread.

What shall i do now?

Best,
Samuel

tuandle, 2019-06-15:
Sorry for the late reply. It's hard to debug what's wrong with your ouster since I never encountered your problem. Maybe you can follow the troubleshooting on page 27 and past the errors here?

erichahn, 2019-06-17:
get_sensor_info
{"prod_line": "OS-1-64", "prod_pn": "840-101396-03", "prod_sn": "991832000853", "base_pn": "000-101323-03", "base_sn": "101831000895", "image_rev": "ousteros-image-prod-aries-v1.12.0-20190502211317", "build_rev": "v1.12.0", "proto_rev": "v1.1.1", "build_date": "2019-05-02T20:17:32Z", "status": "UNCONFIGURED"}
get_alerts
{"alerts": [], "next_cursor": 0}

HappySamuel, 2019-06-18:
Hi @tuandle 

Sorry for the late reply. These are the information that we found:
1) The Hz info is in the screen shots.
[image]

[image]

2) For TCPIP:
a) get_config_txt
{"udp_ip": "192.168.0.125", "udp_port_lidar": 7502, "udp_port_imu": 7503, "timestamp_mode": "TIME_FROM_PTP_1588", "sync_pulse_in_polarity": "ACTIVE_HIGH", "nmea_in_polarity": "ACTIVE_HIGH", "nmea_ignore_valid_char": 0, "nmea_baud_rate": "BAUD_9600", "nmea_leap_seconds": 0, "multipurpose_io_mode": "OFF", "sync_pulse_out_polarity": "ACTIVE_HIGH", "sync_pulse_out_frequency": 1, "sync_pulse_out_angle": 360, "sync_pulse_out_pulse_width": 10, "auto_start_flag": 1, "lidar_mode": "1024x10", "window_rejection_enable": 1}

b) get_alerts
{"alerts": [{"cursor": 0, "code": "ETHERNET_LINK_BAD", "level": "Warning", "realtime": "24582150126", "brief": "Ethernet Link Bad", "description": "Link transitioned to 100/Full which != 1000/Full"}, {"cursor": 1, "code": "ETHERNET_LINK_BAD", "level": "Warning", "realtime": "300166938126", "brief": "Ethernet Link Bad", "description": "Link transitioned to 0/Unknown which != 1000/Full"}], "next_cursor": 2}

c) get_sensor_info
{"prod_line": "OS-1-16-A1", "prod_pn": "840-101396-03", "prod_sn": "991905000414", "base_pn": "000-101323-03", "base_sn": "101839000074", "image_rev": "ousteros-image-prod-aries-v1.12.0-20190502211317", "build_rev": "v1.12.0", "proto_rev": "v1.1.1", "build_date": "2019-05-02T20:17:32Z", "status": "RUNNING"}

Best,
Samuel

tuandle, 2019-06-18:
@erichahn, unfortunately, it seems you need to contact the Ouster support team for returning your unit.
@HappySamuel you have a problem with your ethernet cable as you can see in the `get_alerts`, the ethernet link is not full duplex which prevents the ouster to transmit data efficiently. So you need to fix that.

HappySamuel, 2019-06-18:
Hi @tuandle 

We bought a Gigabyte router and cable and there are no more error but still don't have the Lidar data. Attached the the requested info for the LIDAR but with Gigabyte network. Below are the commands to and feedbacks from LiDAR. What shall we do?

1) get_alerts
{"alerts": [], "next_cursor": 0}
2) get_sensor_info
{"prod_line": "OS-1-16-A1", "prod_pn": "840-101396-03", "prod_sn": "991905000414", "base_pn": "000-101323-03", "base_sn": "101839000074", "image_rev": "ousteros-image-prod-aries-v1.12.0-20190502211317", "build_rev": "v1.12.0", "proto_rev": "v1.1.1", "build_date": "2019-05-02T20:17:32Z", "status": "RUNNING"}
3) get_config_txt
{"udp_ip": "192.168.1.1", "udp_port_lidar": 7502, "udp_port_imu": 7503, "timestamp_mode": "TIME_FROM_PTP_1588", "sync_pulse_in_polarity": "ACTIVE_HIGH", "nmea_in_polarity": "ACTIVE_HIGH", "nmea_ignore_valid_char": 0, "nmea_baud_rate": "BAUD_9600", "nmea_leap_seconds": 0, "multipurpose_io_mode": "OFF", "sync_pulse_out_polarity": "ACTIVE_HIGH", "sync_pulse_out_frequency": 1, "sync_pulse_out_angle": 360, "sync_pulse_out_pulse_width": 10, "auto_start_flag": 1, "lidar_mode": "1024x10", "window_rejection_enable": 1}

Best,
Samuel

erichahn, 2019-06-18:
@tuandle everything was working until I updated to 1.12 from 1.7

tuandle, 2019-06-18:
@erichahn you probably had one of the early hardware version. [This](https://github.com/ouster-lidar/ouster_example/blob/e890d32608ca058da3e2915813e168c93c6f9985/ouster_client/include/ouster/os1.h#L48) says that the current driver supports hardware which is shipped with firmware version 1.9. At this point, I think it's best to contact the Ouster team directly.

erichahn, 2019-06-18:
What's the best contact for technical support, non sales.

HappySamuel, 2019-06-28:
Hi @tuandle 

Thank you for your time, we have sent the malfunction lidar back to ouster, and we got the replacement unit.

Best,
Samuel

closed 2019-06-28 by HappySamuel