ouster-lidar/ouster-sdk #69: IMU and LiDAR axis alignment

chenweng2009, 2019-06-13:
Hi,

I have updated my sensor firmware to the latest version (V1.12.0) and also updated the repository of ouster github but still my LiDAR and the IMU axis do not align with each other anyone facing the same issue ?

Thanks

HappySamuel, 2019-06-14:
I also discover the same issue. The LiDAR axis and IMU axis doesn't align with each other.

Best,
Samuel

tuandle, 2019-06-14:
If you read page 21 in Software user guide v.1.12, you will see that the IMU is slightly offset to the lidar. Read the value from `tf_static` to align them in your software.

Dinatorehanova, 2019-06-14:
Hi @tuandle ,

Thanks for your reply.

Unfortunately, I do not understand the process that you mentioned to align the axis of the (os1_sensor,os1_lidar, and os1_imu).

The software guide mentions about  the following two transformation matrix
M_lidar_to_sensor = [[X, X, X, X], [X, X, X, X], [X, X, X, X], [0, 0, 0, 1]
M_imu_to_sensor = [[X, X, X, X], [X, X, X, X], [X, X, X, X], [0, 0, 0, 1]
But how can I use this matrix to align the axis ?It would be very helpful if you could share the process in a bit detailed way.

Thank you for your time.

chenweng2009, 2019-06-14:
Hi @tuandle ,

Thanks for your reply.

Unfortunately, I do not understand the process that you mentioned to align the axis of the (os1_sensor,os1_lidar, and os1_imu).

The software guide mentions about the following two transformation matrix
M_lidar_to_sensor = [[X, X, X, X], [X, X, X, X], [X, X, X, X], [0, 0, 0, 1]
M_imu_to_sensor = [[X, X, X, X], [X, X, X, X], [X, X, X, X], [0, 0, 0, 1]
But how can I use this matrix to align the axis ?It would be very helpful if you could share the process in a bit detailed way.

Thank you for your time.

chenweng2009, 2019-06-14:
Hi @tuandle ,

A quick followup, If I understand you, I used the tf_static topic value and set them up in my launch file in the form of tf2_ros.But it still does not solve the problem.

<launch>

  <arg name="os1_hostname" default="" doc="hostname or IP in dotted decimal form of the sensor"/>
  <arg name="os1_udp_dest" default="" doc="hostname or IP where the sensor will send data packets"/>
  <arg name="os1_lidar_port" default="7502" doc="port to which the sensor should send lidar data"/>
  <arg name="os1_imu_port" default="7503" doc="port to which the sensor should send imu data"/>
  <arg name="replay" default="false" doc="do not connect to a sensor; expect /os1_node/{lidar,imu}_packets from replay"/>
  <arg name="lidar_mode" default="" doc="resolution and rate: either 512x10, 512x20, 1024x10, 1024x20, or 2048x10"/>
  <arg name="metadata" default="" doc="override default metadata file for replays"/>
  <arg name="viz" default="false" doc="whether to run a simple visualizer"/>
  <arg name="image" default="false" doc="publish range/intensity/noise image topic"/>

  <node pkg="ouster_ros" name="os1_node" type="os1_node" output="screen" required="true">
    <param name="~/lidar_mode" type="string" value="$(arg lidar_mode)"/>
    <param name="~/replay" value="$(arg replay)"/>
    <param name="~/os1_hostname" value="$(arg os1_hostname)"/>
    <param name="~/os1_udp_dest" value="$(arg os1_udp_dest)"/>
    <param name="~/os1_lidar_port" value="$(arg os1_lidar_port)"/>
    <param name="~/os1_imu_port" value="$(arg os1_imu_port)"/>
    <param name="~/metadata" value="$(arg metadata)"/>
  </node>

  <node pkg="ouster_ros" type="os1_cloud_node" name="os1_cloud_node" output="screen" required="true">
    <remap from="~/os1_config" to="/os1_node/os1_config"/>
    <remap from="~/lidar_packets" to="/os1_node/lidar_packets"/>
    <remap from="~/imu_packets" to="/os1_node/imu_packets"/>
  </node>

  <node if="$(arg viz)" pkg="ouster_ros" name="viz_node" type="viz_node" output="screen" required="true">
    <remap from="~/os1_config" to="/os1_node/os1_config"/>
    <remap from="~/points" to="/os1_cloud_node/points"/>
  </node>

  <node if="$(arg image)" pkg="ouster_ros" name="img_node" type="img_node" output="screen" required="true">
    <remap from="~/os1_config" to="/os1_node/os1_config"/>
    <remap from="~/points" to="/os1_cloud_node/points"/>
  </node>
  
  pkg="tf2_ros" type="static_transform_publisher" name="os1_imu_to_os1_sensor" args= "0.006253  -0.011775 0.007645 0.0 0.0 0.0 1.0"/>
  pkg="tf2_ros" type="static_transform_publisher" name="os1_lidar_to_os1_sensor" args= "0.0 0.0 0.03618 0.0 0.0 1.0 0.0"/>
 

</launch>

Thanks for your time

chenweng2009, 2019-06-14:
<launch>

  <arg name="os1_hostname" default="" doc="hostname or IP in dotted decimal form of the sensor"/>
  <arg name="os1_udp_dest" default="" doc="hostname or IP where the sensor will send data packets"/>
  <arg name="os1_lidar_port" default="7502" doc="port to which the sensor should send lidar data"/>
  <arg name="os1_imu_port" default="7503" doc="port to which the sensor should send imu data"/>
  <arg name="replay" default="false" doc="do not connect to a sensor; expect /os1_node/{lidar,imu}_packets from replay"/>
  <arg name="lidar_mode" default="" doc="resolution and rate: either 512x10, 512x20, 1024x10, 1024x20, or 2048x10"/>
  <arg name="metadata" default="" doc="override default metadata file for replays"/>
  <arg name="viz" default="false" doc="whether to run a simple visualizer"/>
  <arg name="image" default="false" doc="publish range/intensity/noise image topic"/>

  <node pkg="ouster_ros" name="os1_node" type="os1_node" output="screen" required="true">
    <param name="~/lidar_mode" type="string" value="$(arg lidar_mode)"/>
    <param name="~/replay" value="$(arg replay)"/>
    <param name="~/os1_hostname" value="$(arg os1_hostname)"/>
    <param name="~/os1_udp_dest" value="$(arg os1_udp_dest)"/>
    <param name="~/os1_lidar_port" value="$(arg os1_lidar_port)"/>
    <param name="~/os1_imu_port" value="$(arg os1_imu_port)"/>
    <param name="~/metadata" value="$(arg metadata)"/>
  </node>

  <node pkg="ouster_ros" type="os1_cloud_node" name="os1_cloud_node" output="screen" required="true">
    <remap from="~/os1_config" to="/os1_node/os1_config"/>
    <remap from="~/lidar_packets" to="/os1_node/lidar_packets"/>
    <remap from="~/imu_packets" to="/os1_node/imu_packets"/>
  </node>

  <node if="$(arg viz)" pkg="ouster_ros" name="viz_node" type="viz_node" output="screen" required="true">
    <remap from="~/os1_config" to="/os1_node/os1_config"/>
    <remap from="~/points" to="/os1_cloud_node/points"/>
  </node>

  <node if="$(arg image)" pkg="ouster_ros" name="img_node" type="img_node" output="screen" required="true">
    <remap from="~/os1_config" to="/os1_node/os1_config"/>
    <remap from="~/points" to="/os1_cloud_node/points"/>
  </node>
  
 I tried to add tf2_ros as static transform publisher in my roslaunch file.But still it does not align.

 pkg="tf2_ros" type="static_transform_publisher" name="os1_sensor_to_os1_imu" args= "0.006253  -0.011775 0.007645 0.0 0.0 0.0 1.0" os1_sensor_to_os1_imu/>
  pkg="tf2_ros" type="static_transform_publisher" name="os1_sensor_to_os1_lidar" args= "0.0 0.0 0.03618 0.0 0.0 1.0 0.0" os1_sensor_to_os1_lidar/>
 

</launch>

tuandle, 2019-06-15:
Sorry for the late reply. I am not sure why you add a static transform. Maybe you want to review [this tutorial](http://wiki.ros.org/tf/Tutorials/Writing%20a%20tf%20listener%20%28C%2B%2B%29). The idea is to transform the lidar frame to the imu frame or to whatever frame that you need.

chenweng2009, 2019-06-15:
Hi @tuandle ,

Thank you very much for sharing the tutorial. Would you mind sharing your code involved in this process(.cpp)?I went through the tutorial that you recommended but unfortunately I was unsuccessful to align the axis.

Thank you very much for your time.

mrgransky, 2020-01-09:
Doesn't the latest Firmware (V1.13) automatically align the transformation between lidar to sensor?
`{"lidar_to_sensor_transform": [-1,0,0,0,0,-1,0,0,0,0,1,36.18,0,0,0,1]}`

and 

`{"imu_to_sensor_transform": [1,0,0,6.253,0,1,0,-11.775,0,0,1,7.645,0,0,0,1]}`

closed 2020-03-11 by dgarbuzov