ouster-lidar/ouster-sdk #73: Need help understanding .bag file message

Justgo13, 2019-06-27:
I currently have a .bag file from recording with the OS1 lidar and I am using ROS melodic on Ubuntu 18.04. Can someone help me understand the numbers in the messages that are being outputted under my topics '/os1_node/imu_packets' and '/os1_node/lidar_packets'. 

This is one of the messages I got from the '/os1_node/imu_packets' topic:
[image]

zuzi-m, 2019-08-22:
The data in the topics is in a form of ROS messages that just wrap the original data from the OS1 sensor in binary format. It is exactly the same data that comes from OS-1 sensor over the network in UDP packets. The format is documented [in the software guide, section 3.4 Lidar Data Format and 3.5 IMU Data Format](https://www.ouster.io/s/software-user-guide-v1120.pdf)

To decode the data you can implement an ROS node that subscribes to these topics and uses the [ouster_client](https://github.com/ouster-lidar/ouster_example/tree/master/ouster_client) library to read the data. An example of a node doing this is the [os1_cloud_node](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/src/os1_cloud_node.cpp) here in this repository.

mrgransky, 2020-01-17:
How about this workaround to get all the topics directly than writing ROS node:

1. Bring up OS1:
`$ roslaunch ouster_ros os1.launch os1_hostname:=os1-xxxxxxxxxxxx.local udp_add:=X.X.X.X lidar_mode:=2048x10 image:=true`

2. Record `my_file.bag` 
`$ rosbag record -a -O my_file`

3. Show info:
`$ rosbag info my_file.bag --freq`

```
path:        my_file.bag
version:     2.0
duration:    30.0s
start:       Jan 08 2020 19:55:54.42 (1578506154.42)
end:         Jan 08 2020 19:56:24.37 (1578506184.37)
size:        2.3 GB            <<<<<<<<<---------------------- My only problem 
messages:    45238
compression: none [894/894 chunks]
types:       ouster_ros/PacketMsg    [4f7b5949e76f86d01e96b0e33ba9b5e3]
             rosgraph_msgs/Log       [acffd30cd6b6de30f120938c17c593fb]
             sensor_msgs/Image       [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/Imu         [6a62c6daae103f4ff57a132d6f95cec2]
             sensor_msgs/PointCloud2 [1158d486dd51d683ce2f1be655c3c181]
             tf2_msgs/TFMessage      [94810edda583a504dfda3829e70d7eec]
topics:      /img_node/intensity_image     297 msgs @  10.0 Hz : sensor_msgs/Image      
             /img_node/noise_image         297 msgs @  10.0 Hz : sensor_msgs/Image      
             /img_node/range_image         297 msgs @  10.0 Hz : sensor_msgs/Image      
             /os1_cloud_node/imu          2978 msgs @ 100.3 Hz : sensor_msgs/Imu        
             /os1_cloud_node/points        298 msgs @   9.9 Hz : sensor_msgs/PointCloud2
             /os1_node/imu_packets        2978 msgs @ 100.1 Hz : ouster_ros/PacketMsg   
             /os1_node/lidar_packets     38083 msgs @ 41.5 kHz : ouster_ros/PacketMsg   
             /rosout                         9 msgs @ 25.0 kHz : rosgraph_msgs/Log       (3 connections)
             /tf_static                      1 msg             : tf2_msgs/TFMessage
```
As you can see, all topics are there, but the size of the bag file is dramatically large for a 30 sec ride :( 
Is there any alternative to address this problem?

closed 2020-01-19 by Justgo13