ouster-lidar/ouster-sdk #74: [Suggestion] use ROS timestamp to publish cloud and imu topic

HappySamuel, 2019-06-28:
Hi

It's a suggestion for Ouster, to use ROS timestamp for both functions that do the topics publishing, packet_to_imu_msg() and cloud_to_cloud_msg(). Otherwise, there will be issue popping out for visualizing pointcloud stream on RVIZ. 

For example i have tf tree, odom -> base_link -> os1_sensor -> os1_lidar. 
**Static Transform Publisher**
base_link -> os1_sensor
os1_sensor -> os1_lidar
**Dynamic Transform Publisher**
odom -> base_link

The issue is raised due to **lookup would require extrapolation into the past. Requested time 3484.289833850 (sensor own timestamp) but the earliest data is at time 1560843465.275178432 (ROS timestamp), when looking up transform from frame [os1_lidar] to frame [odom].** Attached picture is the issue:
[image]

Modification to os1_ros.cpp is as below:
[image]

After this modification, there will be no issue like above. Hope Ouster (master branch) can do modification to use ROS timestamp, instead of using sensor own timestamp.

Hope this help ouster_ros become better. Thank you.

Best,
Samuel

fabioruetz, 2019-06-28:
Hey Samuel,

not from the ouster team but have done some experience with lidars in general.

What you are doing is manually re-stamping the message of the lidar/imu data at arrival time in the ROS node. With this you will introduce a time-offset between when your measurement was actually taken and when you are publishing it. I assume you are seeing misaligned point clouds in you reference frame during fast movements. 

The ouster sensor data is hardware timestamped to a reference clock. From the screenshot it seems that you are running the ouster  in "TIME_FROM_INTERNAL_OSC" mode. This means that at sensor startup the internal oscillator of the ouster will count up and timestamp your sensor data accordingly and hence the big time stamp difference between your ROS and  measurement times. In essence you have not time-synced the ouster to your host pc.

To sync your ouster to you system time of the host computer.  You need to follow 2 steps for this. 
1. Setup a DHCP server on you host computer (Ouster Software-Guide v1.12 ~p 6.). 
2. Enable the ptp sync between your host and the ouster (client).  (Ouster Software-Guide v1.12 ~p 22 ). 
 I hoped this helperd. 

 Hope that helped.
Cheers

HappySamuel, 2019-06-28:
Hi @fabioruetz 

Thanks for the information, i will try it out, syncing the ouster with my system time. I will get back to you once i try the solution you suggested.

Best,
Samuel

fabioruetz, 2019-07-16:
Hi @HappySamuel,
has the issue been resolved?

HappySamuel, 2019-07-22:
Hi @fabioruetz 

I haven't try it out. Will try it out in these 2 weeks. Then i will let you know my status with the ouster.

Best,
Samuel

HappySamuel, 2019-08-05:
Hi @fabioruetz 

I did as you said. Turn it into TIME_FROM_PTP_1588, below are the commands i did:
(1) nc os1-9919xxxxxxxx.local 7501
(2) get_config_txt
(3) set_config_param timestamp_mode TIME_FROM_PTP_1588
(4) reinitialize
(5) get_config_txt
Everything went according to the Software Manual. However, when i run it and with my localization package, the above errors still pop out. 

Please advise which part i missed?

Best,
Samuel

varandaas, 2019-09-10:
Great hint @fabioruetz! We got it working on Ubuntu 16.04 by following these steps:

- Check if your ethernet port supports PTP Hardware Clock: `ethtool -T <eth_port>`
- If so, install `linuxptp`:
    - `sudo apt install linuxptp` 
- Launch the PTP server on the chosen interface (you may need to increase the `tx_timestamp_timeout` by setting a config file for ptp4l): 
    - `sudo ptp4l -i <eth_port> -m -f os1.conf` 
- After setting up the DHCP server on the <eth_port> of the host computer:
    1. nc <lidar_ip> 7501
    2. set_config_param timestamp_mode TIME_FROM_PTP_1588
    3. reinitialize

HappySamuel, 2019-09-11:
Hi @varandaas 

The steps you are doing will enable the ouster lidar to use the ROS timestamp???

Best,
Samuel

varandaas, 2019-09-11:
That's correct @HappySamuel! When you run the ptp4l PTP server (with the -m flag) you will see the ouster trying to sync its clock when you turn it on. These instructions may not be perfect yet, but seem the way to go

mrgransky, 2020-01-14:
@varandaas Thanks for such a descriptive help!
But according to the ouster SW guide, these steps you mentioned are not complete yet, are they?
Because, `phc2sys`, `phc2shm` and `chrony` still need to be configured! Or am I missing something?

closed 2020-03-11 by dgarbuzov

HappySamuel, 2021-02-24:
Hi @varandaas 

Thank you for the info. It's very helpful.

`sudo ptp4l -i <eth_port> -m -f os1.conf`
Besides, i have another question. To run ptp4l, we need to run with sudo. Is there a way that i can place sudo password in the shell script and enable it auto sudo without prompting for password?

Best,
Samuel

learnhard22, 2022-08-20:
> Great hint @fabioruetz! We got it working on Ubuntu 16.04 by following these steps:
> 
> * Check if your ethernet port supports PTP Hardware Clock: `ethtool -T <eth_port>`
> * If so, install `linuxptp`:
>   
>   * `sudo apt install linuxptp`
> * Launch the PTP server on the chosen interface (you may need to increase the `tx_timestamp_timeout` by setting a config file for ptp4l):
>   
>   * `sudo ptp4l -i <eth_port> -m -f os1.conf`
>   * os1.conf:
> 
> ```
> [global] 
> tx_timestamp_timeout 10
> ```
> 
> * After setting up the DHCP server on the <eth_port> of the host computer:
>   
>   1. nc <lidar_ip> 7501
>   2. set_config_param timestamp_mode TIME_FROM_PTP_1588
>   3. reinitialize

Hello @varandaas,
I want to try your procedure on my Ubuntu 20.04 system. I am using an OS0-128 for point cloud data and an EllipseD (SBG Systems) for IMU data in hope to get running a slam package. The IMU is already using ROS time and I found this issue here to set the lidar to the ROS time as well that the point cloud data and the imu data have the same timestamp at the end.
I have a few noob questions, because I am very new into that stuff.
First of all, my ethernet port supports PTP Hardware Clock.

1. How can I launch the PTP server on the chosen interface as you described in your third bullet point?
I think the fact that I am using an OS0 that I have to use the command "os.conf" instead of "os1.."
2. Setting the DHCP server on the ethernet port of the host computer will be done via network settings in my Linux system, right?

I hope anyone could help me.

Best regards,
Lennard

Samahu, 2022-08-24:
A new release has been published which implements the feature discussed in this thread. Please refer to https://github.com/ouster-lidar/ouster_example/pull/420 and https://github.com/ouster-lidar/ouster_example/pull/421 for details