ouster-lidar/ouster-sdk #76: Synchronize Jetson Xavier and Ouster Lidar

SamChuah, 2019-07-02:
Hi, 
I am trying to collect some data using the NVIDIA Jetson Xavier. The Xavier has NIC that can be the master clock for PTP protocol. I am just wondering to synchronize between these two devices, do I just have to simply reconfigure the timestamp mode on Ouster Lidar to PTP and start the ptp4l before I running the lidar and data collection? Or there are some additional setups that need to be done?

constructive-realities, 2019-09-24:
Yes essentially.  You do need to bind it to the right interface... you can also run it as a service to make it zero effort each time you use the lidar.

mrgransky, 2020-01-09:
What does 
> binding to the right interface 

exactly mean?
Do you mean physical connection to the `multipurpose_io` port on the interface box?

closed 2020-12-08 by dgarbuzov