ouster-lidar/ouster-sdk #77: Accessing the ring number

AlexKaravaev, 2019-07-04:
Hi!
I have a problem with organizing ouster lidar into rings. Based on the information contained in PontCloud2 message I have wrote custom point type for PCL. Here is the code:
```c++
#include <pcl/point_types.h>

struct EIGEN_ALIGN16 OusterPoint
{
  PCL_ADD_POINT4D;
  float      intensity;
  uint32_t   t;
  uint16_t   reflectivity;
  uint8_t    ring;
  uint16_t   noise;
  uint32_t   range;
  

  EIGEN_MAKE_ALIGNED_OPERATOR_NEW
};

POINT_CLOUD_REGISTER_POINT_STRUCT  (OusterPoint,
                                   (float, x, x)
                                   (float, y, y)
                                   (float, z, z)
                                   (float, intensity, intensity)
                                   (uint32_t,  t, t)
                                   (uint16_t,  reflectivity, reflectivity)
                                   (uint8_t,   ring, ring)
                                   (uint16_t , noise, noise)
                                   (uint32_t,  range, range)
)
```

However, when I am displaying the ring number of each point in cloud, I get consistently values from 0 to 63. But the model of the lidar, that I am using is OS1-16 with 16 channels, so the ring number should be in range from 0 to 15, if I do understand things correctly.
What are the things, that I am missing?

asherikov, 2019-07-11:
Must be the same as https://github.com/ouster-lidar/ouster_example/issues/51

AlexKaravaev, 2019-07-12:
Same as #51

closed 2019-07-12 by AlexKaravaev