ouster-lidar/ouster-sdk #82: Georeferencing Point Cloud

Ashik19, 2019-07-24:
Hi,

Can anyone guide me with how I can georeference the point cloud so that I can overlap with the real world place? I have the vehicle trajectory gps data but I want to georeference the point cloud.

Thanks

dsb6063, 2019-07-27:
The integrations that use GPS coordinates for sonar 3D modeling would be a good place to start searching. The typical setup involves the high accurate IMU ($8,500) Eclipse 2 Model for example can be used directly with a RTK setup onsite with your base station, or network solution post processed. Either way it outputs the data stream in the marine format used in that field. 

As a result, you would be receiving the  IMU data along with a GPS coordinates timestamped and possibly with post processing 2-10 cm horizontal. What now? This is where you have a manufactured solution where they have done the heavy coding solution (ie OTS) solution. It's does exist. Otherwise, you will need to process the lidar with your own coding solution. I'm certain there is another way; however, the 3D solutions being used in the marine industry are very similar in the setup.   

A less elegant way is to obtain point cloud. 
Convert it to a 3D mesh. 
Clip unwanted data.
Insert it into a software program and manually move it to your GPS alignment.

Ashik19, 2019-07-28:
Hi @dsb6063,

Thank you very much for your reply. I am using an inertial sense sensor which gives both the gps and the imu data.From the gps data I can get the vehicle trajectory but I need to georeference the surrounding cloud according to their real world location. It would be very helpful for me if you could take a moment to lookj into the herewith attach file and guide me on the georeferencing issue.

https://drive.google.com/file/d/1VVpQF4sEs0ZHGyVjx286wNJSiwPKxf6C/view?usp=sharing -INS
https://drive.google.com/file/d/18q4lMwBII0Gln63_W8qPw9Lylm60qjrV/view?usp=sharing-GPS
https://drive.google.com/file/d/10vjUCssLZluBXot2g9_1BwfBVa9FWg0m/view?usp=sharing- Point Cloud

Thanks 
Ashik

dsb6063, 2019-07-28:
If you use Google Cartographer to process your rosbag, then you can integrate the sensor data (INS, GNSS). However, the format of the GNSS is located here 

 

http://docs.ros.org/melodic/api/robot_localization/html/index.html

 

The GNSS format

 

http://docs.ros.org/api/sensor_msgs/html/msg/NavSatFix.html

 

Write a publisher / subscriber to read the input and write out as a rosbag with the appropriate message “sensor_msg/NavSatFix”. 

 

Then, you can use cartographer to SLAM your data with the GPS sensor information. However, it will improve drift, but I didn’t see whether it publishes in georeferenced coordinates. You can post this on the forum on Git for cartographer_ros. 

 

Below is an example input text file for a publisher.

 

https://answers.ros.org/question/236117/how-to-read-a-text-file-and-publish-the-data/

 

I don’t see any public available sources. They are now commercial software applications.

 

 

 

 

 

 

 

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

Ashik19, 2019-07-28:
Hi @dsb6063 ,

Thanks for your reply. Can you suggest me how to publish the timestamp with the pcd file. When converting the bag file to pcd using the (point cloud to pcd input=topic name), the pcd file that I get it does not contain any time stamp.

Thanks

closed 2019-08-17 by Ashik19