ouster-lidar/ouster-sdk #85: Running on Ubuntu Server? No Viz

BMannell, 2019-07-25:
I'm trying to use the Ouster with a Raspberry Pi running a server version of Ubuntu (terminal only) but I run into errors when trying to compile the ouster_ros example. The errors always happen when trying to compile the ouster_viz package. How simple would it be to create a version of the ouster_ros example without the viz capabilities? Is it enough to just remove all reference of viz from the CMakeLists.txt file?

dsb6063, 2019-07-25:
There was a post on this topic. You are correct. You have to remove the references to the rviz in the catkin package builder and cmake. There maybe robot packages on git that you may follow. Try the robots described here: wilselby.com/blog.

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

daniel-dsouza, 2019-08-28:
you can also try my fork which removes the visualization package and its dependencies, among other numerous changes.

jo-ruch, 2019-10-10:
You can try to build only the packages you want with `catkin_make --pkg ouster_ros`

closed 2020-03-11 by dgarbuzov