ouster-lidar/ouster-sdk #86: Timestamp with the PCD files

Ashik19, 2019-07-28:
Hi,

Can anyone suggest me how I can retrive the timestamp when converting the point cloud topic into pcd files using the pcd library (pcl_ros pointcloud  to pcd ) ? It seems the timestamp field is missing in the pcd file .

Thanks

dsb6063, 2019-07-28:
If you don’t see a option to include other fields, then you will need to download that source, modify the fields to include the timestamp. If not directly, maybe place or append it in a descriptor field as a string. Build the code and run it.  Then, parse out the timestamp on the back end in the text file.  

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

Ashik19, 2019-07-28:
Hi @dsb6063 ,
Thank you very much for your kind reply.

I have managed to retrive the timestamp of the point cloud in the .csv file format. Now, I want to convert it into pcd 
https://drive.google.com/file/d/1K6AuJSzY3hAbc3xQ0vwlV1RU3CLrg1Sh/view?fbclid=IwAR0W1xGbg7lqxDr-45q4P6cLGyFm2-bzhpJCQw0quPG9D70FwWJh7AyoL3Y

Now I want to convert it into the pcd file format. Can you recommend me any tool that can convert .csv into .pcd or any workaround method?

Thanks

dsb6063, 2019-07-28:
Cloud Compare. It’s open source. 

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

Ashik19, 2019-07-28:
@dsb6063 ,

I tried with cloudcompare but the problem is that since the csv file does not have the x,y,z value it can not  create the proper pcd file. The problem is if I get the csv fiel I do not get the coordinates of the cloud, I can get the timestamps. But if I convert the bag file to the pcd I do not get the timestamps.

dsb6063, 2019-07-28:
From reviewing the bag file all the data is contained within. rqt_bag shows all the data. You could  "ballpark" the timestamps for an approximate extraction for coordinates. Also, you have all the data to run SLAM on cartographer using the sat_nav option for localization.

Ashik19, 2019-07-28:
Hi @dsb6063 ,

Thank you very much for your kind suggestion. But I guess google cartographer needs very intensive data to run the SLAM and I am using only a 16 beams LiDAR. 

I am just looking for a way to integrate the timestamp with the pcd file generated from the ros bag so that I can later geo-reference the cloud with the timestamps associated with my gps data.

Thanks

mohit-khosti, 2019-07-29:
Hi @dsb6063 
You have a great knowledge about Ouster LiDAR sensors, so it would be very nice of you to join our Ouster forum https://forum.ouster.at/ where you can contribute to this newly created support facility for Ouster LiDAR!

closed 2019-08-22 by Ashik19

Delbina, 2020-12-10:
Hi

How do i change the timestamp of lidar and GPS into DateTime or duration scalars??

my timestamp is like this:  1607426698680700000