ouster-lidar/ouster-sdk #87: Too low z value

Ashik19, 2019-07-28:
Hi,
Anyone facing issue like me with the z value ?

I am getting a very small range of z values (0.69 to -2.12) in one of my files? Can anyone please explain why my z value is too low.I did this survey in a highway environment.

[image]

Thanks

dsb6063, 2019-07-29:
I am interested in your setup. We are doing the same over here with two OS-1 16. What are the ranges? If you are on asphalt, then the returns farther away may  have been affected.  

Were you able to manually registered / align the cloud to the survey data in the horizontal plane. 

Also, we are assuming the GPS vertical has been checked along the road.

The AGL (height of mount) and angle of mount is important for optimal performance SNR.  

Range 

Resolution1.2 cm

Range Repeatability

SNR = 100: ± 3 cm - 1 sigma

SNR = 12: ± 10 cm - 1 sigma

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

Ashik19, 2019-07-30:
Hi @dsb6063 ,

I think the range for os-1 16 is 105 m with 80% reflectivity and 40m with (10%).  

Were you able to manually registered / align the cloud to the survey data in the horizontal plane?

I think you mean SLAM right? the ouster LiDAR unit automatically  align the point clouds which inhibits the point overlaps. I found if you go with a single unit like me at a highway speed you tend to miss a lot of objects which is typical i guess considering only 16 beam and ouster does not provide multi-return. However, I did understand the process you mentioned about the sound-noise ratio. What are the ranges of Z value in your cloud ?

Thanks

dsb6063, 2019-07-30:
I am viewing your screen capture. Are the value consistent in the their relative values? The question sounds like if I start at elevation 100 meters on the highway, why are the lidar elevations not positive. I am not certain of the initial value of the calculations. It could be zero based on the sensor offset values. Then, negative values would be expected. 

If the data is good, then you can move it to your survey control. If you have no survey control, that would be an issue to rule out any loss of a fixed solution. Survey control validates input from the GPS is matching field values. 

I will take a look at the GPS time for you. I hope this email helps, but it sounds like the starting elevation is basically zero.

Ashik19, 2019-07-31:
Hi @dsb6063 ,

I just checked all of my files again it seems the z value is consistent. But all of them has negative z value with maximum positive z value of (2m) which I do not understand.The data is collected from a highway environment.  Would you mind sharing the z values from your data ?

Thanks

dsb6063, 2019-07-31:
Using traditional laser scan data, the output generates a scan world. A scan world is an arbitrary reference point and coordinate system of the point cloud. It appears your data is the same in that regards. It is in it's own scan world and has not been referenced to a known dataset; thus, creating a georeferenced point cloud. The question that I asked earlier is how you are going to drag and drop that point cloud to a known coordinate system. 

How I would approach this is have targets in the field (3). Survey the targets manually, Overlay (ie localize or translate and rotate) the scan data to the survey points. What software? There are several that can do this heavy transform. Cloud Compare, ReCAP to Infraworks. On GIS side, I see that you are using Arc Scene. Document what you do and how was the result.

Ashik19, 2019-08-01:
Hi @dsb6063 ,
Got the error. I had a wrong understanding. The (x,y,z) coming out of the ouster are in degree. The negative z value resembles the points which are below the horizontal plane.What are the values that you get from the sensor ? Apart from (x,y,z and intensity)

[image]

Thanks
Ashik

closed 2020-03-11 by dgarbuzov