ouster-lidar/ouster-sdk #88: Syncing GPS Pulse with the LiDAR Unit

Dinatorehanova, 2019-07-30:
Hi, 

Can anyone please guide me how I can sync the GPS time with the point cloud topic? I have checked the ouster LiDAR FAQ side it mentions there is a dedicated pin in the LiDAR unit which accepts the GPS pulse

https://www.ouster.io/faq

Has anyone tried it? 

Thanks

dsb6063, 2019-07-30:
The pps master pin in the black box is actually the NEMA input port for reading NEMA sentences. The other one is PPS slave  (pulse per second). Supported Baud rates are limited. 9600 is supported. If you purchase a lower cost GPS for timing, then make sure it supports 9600 baud. 

 

You need to setup PTP on your computers, if you decide not to use GPS time described above. That will sync the linux or computer time depending how its setup.

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

Dinatorehanova, 2019-07-30:
Hi @dsb6063 ,

Okay I got your point. But can you tell me how I can get the timestamps of my point cloud topic so that I can match it with the gps data?

Thanks

dsb6063, 2019-07-30:
But can you tell me how I can get the timestamps of my point cloud topic so that I can match it with the gps data?

Once it is recorded in a rosbag, you can play and record, but you need to have a transform written. Basically, publisher and subscriber. That is more work. I would modify the code in the ros cookbook. 

http://wiki.ros.org/rosbag/Cookbook

Reorder a Bag File Based on Header Timestamps

Change line 46      last_time = time.clock() to be last_time = your GPS Start Time. (probably will require formatting and initializing.)
 
Also, change the line 29 to be a large number of seconds, since this code is meant to reorder packets that are out of order. 

If you have problems, send me the GPS start time, I’ll try it in the morning. 

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

Dinatorehanova, 2019-07-30:
Hi @dsb6063 ,

Thank you very much for your reply. I am quite new to ROS platform. Can you please spare a moment to look into my bag file which contains the point cloud, gps and the imu topic(os1_cloud_node/points, /gps, and /imu). I need to find a way to sync the timestamps of  imu, gps and the point cloud. I have already extracted the gps and the imu topic as csv format from the bag file.  

gps - https://drive.google.com/file/d/1IIWJh9BRIybc_N03vqC6ludGFz42N0e1/view?usp=sharing
imu - https://drive.google.com/file/d/1rjnDJEiC5uYWyyc1ILKgSPX2gsbMuNa7/view?usp=sharing
ros bag - https://drive.google.com/file/d/1NDk4TvB07JMW6kT7vThGoCOx1lgo0soI/view?usp=sharing

Thank you very much for your valuable time. I will wait for your replay.

dsb6063, 2019-07-31:
I did complete the timestamp. It's not used that way since it could destroy the data used in SLAM. I am waiting the the download of your file. If you can debug python, I can just upload the script. Also, the timestamps are based on unix time in your files.

Dinatorehanova, 2019-07-31:
Hi @dsb6063 ,

Thank you for your reply. you have made a script which can extract the timestamp associated with the point cloud topic from the ros bag ? Can you please share the script ?
And can you please explain what do you mean by It's not used that way since it could destroy the data used in SLAM ?

Thanks again for your kind help.

dsb6063, 2019-07-31:
[reorder.zip](https://github.com/ouster-lidar/ouster_example/files/3453752/reorder.zip)

When you adjust the timestamps of the header packet, you are affecting the arrival time of the lidar packets. SLAM algorithms are time dependent.

Dinatorehanova, 2019-08-01:
Hi @dsb6063 ,

Can you please tell me why I am getting the following error?
How do you run the code?

I have changed the rosbag file directory in the code
bagfile = '/home/nvidia/2019-07-09-14-19-30.bag'
[image]

[image]

Thank you very much for your valuable time.

Dina

dsb6063, 2019-08-01:
If that the path to your bagfile, you need to target python 2.7 not 3.0. A few suggestions that may make that debugging easier is either download Eclipse IDE or VSCode IDE. Both are graphical and allow switching python versions.  

Hi @dsb6063 <https://github.com/dsb6063>  ,

Can you please tell me why I am getting the following error?
How do you run the code?

I have changed the rosbag file directory in the code
bagfile = '/home/nvidia/2019-07-09-14-19-30.bag'
 <https://user-images.githubusercontent.com/51459647/62278789-00ef5880-b406-11e9-8bb3-7045caabf0bf.png> 

 <https://user-images.githubusercontent.com/51459647/62278798-06e53980-b406-11e9-98ec-9c73ca433736.png> 

Thank you very much for your valuable time.

Dina

—

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

Dinatorehanova, 2019-08-02:
HI @dsb6063 ,

Would you mind sharing the output of this code please ?

Thanks

dsb6063, 2019-09-09:
I have attached my setup for gps signal and pps setup. For the hardware, I attach the GPS transmit to Master and PPS to the slave on the Ouster Hardware Box. Also, connect the ground to the GND. See the attachment for the network commands. I only used a cheap GPS transmitter for the signal. There are other setup variants. This is one that worked for my setup.

[NMEA-SETUP-OUSTER.pdf](https://github.com/ouster-lidar/ouster_example/files/3588281/NMEA-SETUP-OUSTER.pdf)

Dinatorehanova, 2019-09-09:
@dsb6063 

How did you connect the gps with the port ? I think the pps method needs you to be connected with the interface of ouster so that it gets synced

Dinatorehanova, 2019-09-09:
Hi @dsb6063 

I have a question how you interpolate the GPS data since the LiDAR update rate is very high compared to the GPS. How you you interpolate the GPS data to match with the LiDAR scan rate. The GPS I am using it is updating in every 200 milliseconds. I am trying to geo-reference my point cloud to the corresponding real world place so that it overlaps 

Any help would be great

Dinatorehanova, 2019-09-10:
@dsb6063 

I am also getting the time information of the sensor in the unix time stamp which sync with the processor time. But when I am opening the pointcloud in the csv format (/os1_cloud_node/points topic) the values under the t column does not make sense at all. 

Please have a look at the herewith attached file and picture

[image]

https://drive.google.com/file/d/1g4_VQKwM4TpnDWkZNnO7Vy1GtsdGvRg8/view?usp=sharing

dsb6063, 2019-09-10:
If you are using only one sensor, then no need to correlate GPS data. However, you need to track the GPS time when you want to merge the position information from a GPS, or perform the operation manually. 

 

Below is a set from csv file. 10^14 (units ?). It also appears you are losing information on the export into the csv file. From get_time_info appeared to be correct and the linux time shown was in 10^9 seconds.  What process are you using to create the csv?

 

156282000000000.000000000

156282000000000.000000000

156282000000000.000000000

156282000000000.000000000

156282000000000.000000000

156282000000000.000000000

156282000000000.000000000

156282000000000.000000000

156282000000000.000000000

 

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

dsb6063, 2019-09-10:
1/200 ms = 5 hz  (5 messages /sec). I am not certain on the upper limit for a 9600 baud connection. Therefore, you are left with interpolation of the time. What is the best approach? I do not have a good answer, yet! I boards I reviewed several months ago removed the duplicate timestamps. However, for 16 channel it may have a detrimental impact on the quality. 

The experts should chime in on this topic. One approach that I see is increment the duplicate timestamp by a minimum of 1 nanosecond to correct the duplicates without significantly impacting the data. I have see little information on the rosbag prep for these scenarios. If you find a solution, then post it back here. The ros cookbook has the timestamp code example.  

Another approach is run your data collection lines more than once to collect enough data that a few can be dropped. 

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

dsb6063, 2019-09-10:
Connect the pps from the GPS to Slave pin on the interface module and connect the NMEA port from the GPS (transmit) to the Master (Multipurpose IO).

1.	For the NMEA use, set the Mulitpurpose IO to NMEA input, this turns the port into a input channel that reads the NMEA sentence. 
2.	You can connect the PPS from GPS to Slave pin on the interface for pulse per second updates.
3.	Also, connect the ground from GPS to GND on the interface.

My previous attachment has the network commands.  I did not isolate the PPS command for just PPS updates. 

 

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

Dinatorehanova, 2019-09-10:
Hi @dsb6063 ,

I really do not know how I can connect my gps with the sensor.And I think I would relay on the PTP 1588 because my workstation is somewhere where I do not get the gps connection except going outside.I am trying to stitch the system time on the LiDAR points. When I run the get_time_info it shows me the time of the system. But when I use the pdal to retrieve the PCD file generated from the point cloud to pcd command, it seems the values under the T column is very irrelevant. Please help me with the following queries
1. How did you get the timestamp (156282000000000.000000000) ?
Is it from my data or your data ?
2. When you change the time sync mode from the TCP, do you need to also change this thing into some place of the code in order to make it effective properly ?

Thanks . I will wait for your reply.

dsb6063, 2019-09-10:
Your question.

When you change the time sync mode from the TCP, do you need to also change this thing into some place of the code in order to make it effective properly ?

That is fine. On your GPS unit, you need to find a way to get the transmit, pps, and ground lines. This does require opening the interface box lid that voids your warranty. 

 

A.	Your GPS to Ouster Interface please connect these 3 wires.

 

               Transmit -->PPS_Master (this is the NMEA sentence)

               PPS --> PPS_Slave

               Ground --> GND

 

B.	Now, set the TCP commands to setup and read the  NMEA Sentence (Ouster hardware version 1.12)

 

set_config_param_multipurpose_io_mode INPUT_NMEA_UART

set_config_param timestamp_mode TIME_FROM_SYNC_PULSE_IN

 

May need to set the baud rate. If your baud rate is not 9600, see the Ouster specs.

 

GPS and Interface connection must be set at same baud rate.

 

For great schematics, Velodyne has a few with GPS. You do need to interpret between the two hardware devices. 

 

Your question.

How did you get the timestamp (156282000000000.000000000) ?
Is it from my data or your data ?

When I downloaded your dataset (csv), I opened Excel. I increased the supporting characters to show the data. As you can see, it was truncated. 

 

 

  <https://github.com/notifications/beacon/AGHBSQJ3XBMGBAQCLCPUSJ3QI6M7HA5CNFSM4IHXJJPKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD6LBNXA.gif> 

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

Dinatorehanova, 2019-09-10:
Hi @dsb6063 

Can you please share one of your pcd file here ? 

Thanks

dsb6063, 2019-09-10:
When I converted to the PCD file and then process it to Cloud Compare. I did not use the time stamps. I don’t have any readily available. The test ones are monochrome due the intensity not being recognized. Once I get started again collecting, I will contact you. It should be to long. 

 

 

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

Dinatorehanova, 2019-09-10:
Thanks @dsb6063 

I understand your point. But I just wanted to check the values that you are receiving under the (t) field in your cloud. If you could share one single pcd which you generated after syncing the LiDAR with the gps I can quickly take a look.

Thanks

closed 2020-03-14 by dgarbuzov