ouster-lidar/ouster-sdk #89: Check if Ouster is in sync with Jetson Xavier using ptp4l command

mohit-khosti, 2019-07-30:
I set up my system with Ouster connected to Jetson Xavier and try to sync it by using ptp4l command. However, how can I check if the sensor is in synced?

dsb6063, 2019-07-31:
Run the ouster client. Watch for the timestamp. It may take 1-2min.

Chenweng1, 2019-08-01:
Hi @dsb6063 ,

Can you please tell me what are the full names of the following command and how I can save the data ?

[image]

Thank you very much for your time.

Weng

dsb6063, 2019-08-01:
_ts field is your timestamp. One for  the IMU and other for Sensor. Here is a site to show the current unix timestamp that you would expect to view. https://www.unixtimestamp.com/index.php

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

Chenweng1, 2019-08-01:
Hi @dsb6063 ,

So the first one is the imu timestamp and the third one is the point cloud timestamp ? Can you please tell what are the other 4 columns stand for ?

dsb6063, 2019-08-01:
What you are looking for is bottom of page 18 and 19 showing the data format. This is found in the Software Manual on the downloads page. That will help. 

 

https://www.ouster.io/downloads

 

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

closed 2020-12-08 by dgarbuzov