ouster-lidar/ouster-sdk #91: Ouster 64 cannot detect black squares in checker board

cynosure4sure, 2019-08-03:
Hello!

I have a new ouster OS1-64 beam unit on firmware version 1.12.
I noticed a strange issue where ouster cannot detect objects of black color e.g. I placed some checkerboard patterns in front of the device and in both ouster visualization and rviz it cannot detect black squares. Please see images below for reference:
[image]

[image]

Similarly, in the rviz visualization if you see the intensity bounds in the side panel it shows a larger range as compared to other ouster units we have in the lab. 

My question would be do I have a defective unit? or am I setting some parameter on the device incorrectly?

Thank you.

dsb6063, 2019-08-03:
That is quite interesting. In general, the color black absorbs light; therefore, the texture of the material would be helpful to understand your issue. Is it smooth, shiny and reflective, or dull and rough.  I would like hear the official response. We can usually track black targets with newer lasers but with some difficulty. Shiny black tends to be more difficult from using our equipment; however, rough texture tends to bounce and reflect more light. You may want to try adding texture to square just to see if it reflects any light.  Maybe shiny will work better?

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

cynosure4sure, 2019-08-04:
I contacted Ouster team and they said it is a known issue and they shared v1.13 beta version to resolve the issue. However, I haven't tested the new firmware on my unit yet.

closed 2019-08-04 by cynosure4sure

nicolasrosa, 2019-10-22:
Hello @cynosure4sure, did the firmware update solve the problem?