ouster-lidar/ouster-sdk #93: Coverting .pcap files to .las

mohit-khosti, 2019-08-07:
Hey everyone,
I am using Ouster Studio and it recording the data in .pcap format and I would like to know if there is any way to convert it into a .las format? 
It doesn't matter weither on Windows or Linux!
Thanks

jimycelis, 2020-10-08:
Hi, have you already managed to convert the data from ouster to LAS? Thank you

sam-rke, 2021-01-27:
Hi @k-mohit  and @jimycelis 
Are you guys able to convert from .pcap to .las?
If so kindly enlighten us with how to do that?

kairenw, 2022-02-03:
Leaving a note here that there is an example in our Python SDK which converts to LAS: 
https://static.ouster.dev/sdk-docs/examples.html#converting-pcaps-to-other-formats

Will leave this open as pcap converstion to LAS in cpp is still not addressed.

twslankard, 2023-09-06:
I'm closing this issue due to inactivity.

Anyone wishing to request an enhancement or report issues with the examples should open a new issue.

Best Regards,

Tom Slankard
Staff Engineer
Ouster Inc.

closed 2023-09-06 by twslankard