ouster-lidar/ouster-sdk #94: Difference between (/os1_cloud_node/points and /os1_node/lidar_packets)

Dinatorehanova, 2019-08-09:
Can anyone please help me to understand what is the difference between this two topics ?

- /os1_cloud_node/points

- /os1_node/lidar_packets

And why the (/os1_cloud_node/points) topic bag file consumes a lot of file space ?

Thanks

zuzi-m, 2019-08-22:
`/os1_node/lidar_packets` publishes raw data in the same form as it comes out of OS-1 sensor - each message contains one lidar packet which stores 16 columns of points (for more info [see software guide, section 3.4](https://www.ouster.io/s/software-user-guide-v1120.pdf))

`/os1_cloud_node/points` publishes [sensor_msgs/PointCloud2] data where each message is one full frame of point cloud from OS-1, reconstructed from incoming lidar packets. Each point in this point cloud has its XYZ position, range, intensity, ambient noise, reflectivity etc.

The size difference is mainly because the data in the `lidar_packets` is packed more efficiently in memory and does not contain XYZ positions of the points (they are computed from range). For more detail you can look at [my other reply here](https://github.com/ouster-lidar/ouster_example/issues/90#issuecomment-523829269).

Dinatorehanova, 2019-08-23:
Hi @zuzi-m ,

Thanks for your kind reply. Can you help me with one question ?

When I am converting my the point cloud into the pcd file using the pcl_ros library then I am getting a constant value of 0 under the  the noise,time, ring, reflectivity, range column. And also in terms of the x,y,z many points has a value of 0.

[image]

When I am converting the bag from bag to pcd using the pcl_ros it says for each frame it has the value for (timestamp, range, reflectivity,x,y,z, intensity)

[image]

Thanks

zuzi-m, 2019-08-23:
@Dinatorehanova what type of point for the `pcl::PointCloud<T>` are you using? For the point cloud to work properly and have all the fields you have to use the custom point type defined in [point_os1.h](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/include/ouster_ros/point_os1.h)

Sample code in a callback receiving the point cloud from `/os1_cloud_node/points`:
```
#include <sensor_msgs/PointCloud2.h>
#include <pcl/point_cloud.h>
#include <pcl_conversions/pcl_conversions.h> // pcl::fromROSMsg
#include <ouster_ros/point_os1.h>            // ouster::OS1::PointOS1

void cloud_callback(const sensor_msgs::PointCloud2::ConstPtr& m)
{
  pcl::PointCloud<ouster_ros::OS1::PointOS1> cloud;    // Define a PCL point cloud with correct point type
  pcl::fromROSMsg(*m, cloud);                          // Convert ROS message to PCL
  pcl::io::savePCDFileASCII ("test_pcd.pcd", cloud);   // Save point cloud to file
}
```

Dinatorehanova, 2019-08-24:
@zuzi-m ,
I am directly using the bag file to feed in the pcl_ros library using the command rosrun pcl_ros bag_to_pcd <input_file.bag> <topic> <output_directory>

As topic, I am using the /os1_cloud_node/points.

I am recording the /os1_cloud_node/points topic in my ros bag. Then using "rosrun pcl_ros bag_to_pcd <input_file.bag> <topic> <output_directory>" command to get the pcd file.

Can you please explain what do you mean by using custom point type ?

Ashik19, 2019-08-25:
@zuzi-m ,

We are also facing the exact same problem same as @Dinatorehanova .Converting from rosbag containing (/os1_cloud_node/points) topic into pcd file using the bag to pcd command from the pcl_ros library producing pcd files with constant value of '0' for (t,noise,range,ring,reflectivity) and also too many points have a valeu of (0) for the x,y,z. It would be very helpful if anyone can guide on this issue.

Thanks

dsb6063, 2019-08-25:
There was a discovered firmware issue that was posted last week with gaps in the data. I’m curious if the ‘0’s you’re a seeing are actually related to the firmware issue. If you run your rosbag in rviz, are there any visible missing blocks or gaps in the data? If there is an issue with the values in the rosbag, then any conversion will result in 0s or the default value for empty set. 

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

Ashik19, 2019-08-25:
@dsb6063 ,

Thanks for your kind reply. I have updated the firmware to the latest one V 1.12 but still it's coming up with the same problem. May I know whether you also receive the same error Like a constant value of 0 for the (range,reflectivity,ring,noise,timestamp) when you are converting the bag file into pcd using the pcl_ros ?

Thanks

closed 2020-03-14 by dgarbuzov