ouster-lidar/ouster-sdk #95: Data block unpacking

NizarTarabay, 2019-08-13:
Hello,
I am currently trying to unpack UDP lidar data to a readable format. I am using the struct.unpack function, which works fine with the IMU data; however, I am not able to unpack the data block in the Lidar data packet (range, reflectivity, signal photons, and ambient noise).  
Do you have any suggestions on how to unpack the data block in the UDP Lidar data to a readable format? and what is the data type of the range, reflectivity, signal photons, and ambient noise?
Thank you!

dsb6063, 2019-08-13:
1.	You can use Wireshark to view the individual packets. 
2.	Use Ouster Client to view the live data from the sensor. https://github.com/ouster-lidar/ouster_example
3.	Use the https://www.ouster.io/downloads and get the software guide to view the packet data structure.

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

NizarTarabay, 2019-08-13:
Thank you for your reply,
I have no problem in viewing individual packets. The problem is how to interpret them. the documentation (software guide) provide a clear description on the data structure of the IMU and the Lidar data but only for the timestamp, measurement ID, frame ID, encoder count, and packet status (I have no problem in unpacking and reading them). The documentation does not provide sufficient information on the range, reflectivity, signal photons, and noise photons **datatype**. 
Moreover, while reading data from ouster 16 channels, it is not clear in **which set of words (word 4 -> word 195) the lidar data blocks are written**? 
Any help would be appreciated
Thank you!

Ashik19, 2019-08-15:
Hi @NizarTarabay ,

Can you please share your working process on how to unpack the lidar data packet from ros bag? I am using the pcl ros library to convert my rosbag into pcd files but my pcd files only contains the x,y,z and the intensity value for each of my points while the 't', 'reflectivity', 'noise', 'ambient' and 'range' has a value of 0. If you could share your working process to extract these information from lidar packets that would be great.

Thanks

NizarTarabay, 2019-08-15:
Hi @Ashik19,

I am not extracting data from the ros bag file, I am extracting them as UDP package directly from the network by using **socket programming**. In this way you get access directly to the raw data streamed by the lidar (not sure if this is the optimal solution to do that :) ) any suggestion would be appreciated. In my case I am still not able to extract the 'reflectivity', 'noise', 'ambient' and 'range' data.
x, y, and z coordinates are computed using the **range**, encoder count and beam azimuth angles, reference to the software guide for the formulas.
I believe that there is a relation between intensity and reflectivity/range/ambient. Not sure about that.
I don't have experience with pcd files, but I recommend you to go to the [documentation ](http://pointclouds.org/documentation/tutorials/pcd_file_format.php) to check their format and why you are only seeing xyz and intensity data.

I hope this will help!
Best

dsb6063, 2019-08-15:
This may not be the layer abstraction you are seeking. You can adjust the client headers to print what you want to view.

https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/src/main.cpp

A description found here shows how to access the values.

   std::transform(
            cloud.begin(), cloud.end(), ls->begin(), [](const PointOS1& p) {
                return ouster::LidarScan::make_val(p.x, p.y, p.z, p.intensity,
                                                   p.t, p.reflectivity, p.ring,
                                                   p.noise, p.range);

https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/src/viz_node.cpp

This would require setting up an IDE and debug environment. If you are at the UDP abstraction layer, then this may not be the answer.

zuzi-m, 2019-08-16:
@NizarTarabay I had the same question when I tried implementing the same as you - extracting the point values from a UDP packet sent by OS1-16 sensor. The data format is documented in the software guide, section 3.4 Lidar Data Format.
According to my experience it is the following (for both 16 and 64 channel sensors):
- each packet (12608 bytes) contains 16 columns of data
- each column is 788 bytes long and consists of a header, points data, and a status value
  - Header: contains timestamp, measurement id, frame id and encoder count (16 bytes, used to identify which column of points it is in the point cloud)
  - then the column continues by 768 bytes of data which makes space for 64 **data blocks** for the points in corresponding column (768 bytes = 192 words = 64 * 3 words per point)
  - each column ends with 4 bytes of data - packet status indicating if the data is valid

The tricky part is that if you use the 16-channel sensor, only 16 points in each column will be valid. I could not find any information in the documentation specifying which 16 data blocks out of the 64 will be used, but when looking in the raw data, I saw that my sensor uses blocks with index 2, 6, 10 ... (starting with the **third** block and then every 4th block). Other blocks are set to zero (which is another way of selecting only valid points - only points with non-zero range are valid)

As for the structure and data types of values for each data block for points, you can find that in documentation too, it's the second table in section 3.4, titled `Full Description of Data Block:`

Hope this helps! [You can have a look at the example source codes to see how the data is extracted from the UDP packet here](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/include/ouster/os1_util.h#L132-L137)

Ashik19, 2019-08-16:
Hi @zuzi-m ,

Can you share the process on how to apply the source code on a rosbag (lidar_packet) to extract the data ? I am saving my data in rosbag format but I have not been able to read the packet data.

Thanks

zuzi-m, 2019-08-19:
@Ashik19 the process for reading the rosbag packets is the same as reading the UDP packets from the network because the Ouster's [os1_node](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/src/os1_node.cpp) only reads the UDP packet to a buffer and publishes the data in the `PacketMsg` ROS message.

To read the data, subscribe to the topic publishing these packet messages, and in the callback you can access the data by `message.buf.data()`. You can look at the source code of the [os1_cloud_node](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/src/os1_cloud_node.cpp) to see how you can use ouster's code to read the packet into a point cloud (though it is a bit tricky to understand because it uses lambda functions and iterators).

[This piece of code](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/include/ouster/os1_util.h#L91-L140) shows how to read raw packet data (in the `packet_buf` buffer that you would get from reading the UDP packet or the ROS message data) and get all the values from it.

Ashik19, 2019-08-19:
Hi @zuzi-m,
Can you please share the process to apply the function (os1_util.h) on the packet data in a bag file? I am playing my bag file containing lidar packets. But can’t understand how to implement the function on my file. I am following the steps like 

rosbag play file.bag

rostopic echo /os1_node/lidar_packets 

Then it’s printing the file contents. But don’t understand how I can feed my packet on the function so that it returns data( timestamp, ambuent,range,measurement id, frame id etc.) in the human readable format

Chenweng1, 2019-08-22:
@zuzi-m ,
 I will comment on this as it is still open. the ouster manual does not provide any specific information about how to extract the data from the ros bag file and also about the data type of range, ambient, reflectivity. I assume, the os1_util.h file feeds the /os1_node/lidar_packets topic and gives the output.You get 16 timestamps for each packet.Anyone has any idea about the process of applying the os1_util.h on the ros bag file ? Can anyone share the command line ?or screenshot will be very helpful.

Thanks

zuzi-m, 2019-08-22:
As I mentioned here earlier, the ouster manual does describe the data format of each UDP packet very well in their software manual, in section **3.4 Lidar Data Format** ([page 18 in this document](https://www.ouster.io/s/software-user-guide-v1120.pdf)).

The data in rosbag files is in a form of custom ROS messages defined in [ouster_ros ROS package](https://github.com/ouster-lidar/ouster_example/tree/master/ouster_ros), which are created by the os1_node and each message contains a data buffer holding exactly the same data as the original UDP packet, so the way to extract data from this message is the same as reading UDP packets.

The way you can use a sample rosbag file with this repository is also very well documented, in the README.md files in this repository, and each of the three packages inside it. Please have a look at [ouster_ros package](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/README.md) that provides clear instructions on how to run a rosbag file using this code with ROS.

Chenweng1, 2019-08-22:
Hi @zuzi,

Thanks for your reply. My question was how to extract the range,noise from the bag file. When I use the command rostopic echo /os1_node/lidar_packets -b file.bag -p>name.csv. It creates a csv file with timestamp for every single message. And for every single timestamp I get 12608 buf number. I do not understand as you mentioned in your previous comment like how to apply the os1_util.h file on my ros bag so that I can retrieve all the data in human readable format. I would really appreciate if you could share the command line or the screenshot for this process.

Thanks

Chenweng1, 2019-08-22:
Once you record a bag file, how you apply the os1_util.h file on your packet message (/os1_node/lidar_packets) so that it gives the timestamp, range, x,y,z, ambient ? The os1_util.h being a header file it can not be executed like .cpp right ? But that’s the file we need to apply on recorded data to get the information. Any help would be great.

zuzi-m, 2019-08-22:
@Chenweng1 the data is in binary format so if you print it the way you are trying, you will not be able to see anything.
To "apply" the given source code on the file, you have to implement your own ROS node, include the `ouster_client` library, subscribe to the `/os1_node/lidar_packets` and use the library in your code to parse the packets. That is exactly what [os1_cloud_node](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/src/os1_cloud_node.cpp) does so you can use that as an example.

ethan20120, 2019-08-23:
Hi @NizarTarabay  @zuzi-m,

How are you stitching the packet data with your point cloud? what's the file format you are using to save these unblocked packet data ? Anyone who is using rosbag to save the cloud in the pcd format I believe are saving the point clouds in the pcd format which only supports the x,y,z, and the intensity attribute not the other attributes.

Dinatorehanova, 2019-08-23:
Can anyone please share a script to extract the information from the ros message (/os1_node/lidar_packets)?  Maybe someone from ouster can publish a script which will produce human readable data as many of us do not have proficiency in programming. we are saving data in rosbag format. But unfortunately, yet could not parse the attribute of the point clouds. I will really appreciate any help.

Best
Dina

dgarbuzov, 2020-12-07:
We currently provide sample code to dump point clouds to csv here: ouster_client/src/example.cpp. For reading data from a bag, you could try echoing the *point cloud topic* not the raw packets.

Your best option to get human-readable data without requiring any programming might be to use Ouster Studio, available for download here: https://ouster.com/downloads/. It includes tools for outputting CSVs from recorded pcap data.

closed 2020-12-07 by dgarbuzov