ouster-lidar/ouster-sdk #96: Missing range data

NizarTarabay, 2019-08-14:
Hello,
The range image data as well as the point cloud are significantly missing points as shown in the picture below. this problem occur in an irregular manner. Once the Lidar starts, the range image is clear, after a short time the range points starts to disappear, howver, the intensity data are quite clear.
Does anyone face this problem and what is the solution

dsb6063, 2019-08-16:
If the intensity map is clear, then it sounds like a software graphics issue. Is this ouster studio? If it is, you may search the kitware.com boards and support.

NizarTarabay, 2019-08-16:
It's a software issue, going to be fixed in the next firmware update.
I am using the ros visualizer. 
Thanks!

closed 2019-08-16 by NizarTarabay

zuzi-m, 2019-08-16:
@NizarTarabay could you please share some link to the information you found about this? I encountered the same problem. Is it an issue with the sensor firmware or just the visualization?

NizarTarabay, 2019-08-16:
Hi @zuzi-m 
It is a firmware issue, according to the Ouster team. Try to contact them and they will guide you through the solution, they may give you an update to a new firmware. 
In my case a firmware update solve this issue.
Thanks!

emzhang, 2019-08-20:
Indeed a firmware issue. If anyone else is experiencing this, please reach out to our team at support@ouster.io and we can help resolve it.

jhultman, 2019-12-27:
@emzhang Would this firmware issue explain the following "propeller effect"  we are seeing on our OS1 16-beam lidar? There is nothing obstructing the sensor but some parts of the field of view have blind spots.

[image]

NizarTarabay, 2019-12-27:
Hello @jhultman,
You are probably seeing the "propeller effect" because the lidar uses the udp protocol to communicate with your computer (udp does not guarentee a lossless data transmission). What you are seeing here is most likely a loss in udp packets. Try to increase the udp buffer size in your computer.

jhultman, 2020-01-03:
@NizarTarabay Thank you for your suggestion. I tried increasing UDP buffer size using `sysctl -w net.core.rmem_max=26214400` but continue to experience the same problem. Do you have any other suggestions? Otherwise I may contact Ouster and see if they suggest firmware update. Thanks!

NizarTarabay, 2020-01-07:
@jhultman, To overcome this issue I wrote my own program that saves the UDP packet into a file, then preprocessing/visualizing them. you can check the code [here](https://github.com/NizarTarabay/Extract_packets_os16).
What I noticed that even with the 16 channels lidar, the packets are always equal to 12608 Bytes, same as for 64 channels. What I did in my code is removing the unwanted packets that belong to unused channels so I managed to reduce the packet length to 3152 Bytes.

HappySamuel, 2020-02-11:
Hi @NizarTarabay 

When you said about the firmware update, do you refer to upgrading to firmware v1.13 ? For you to reduce the packet length to 3152 bytes, is that for os1-16?

Best,
Samuel

NizarTarabay, 2020-02-11:
Hi @HappySamuel 

- I believe it was the v1.12. I'm not sure though.
- Yes, reducing the packet size is only for the 16 channel lidar.

Best,
Nizar

HappySamuel, 2020-02-11:
Hi @NizarTarabay 

How shall I do with the ouster code to filter out those 48 channels (excess with zeros) before publishing it into ros topic? Need some guidance on where to edit.

Best,
Samuel