ouster-lidar/ouster-sdk #98: Software timestamping with ptp4l

Ashik19, 2019-08-15:
HI, I have set my lidar timestamp mode to "Time_From_PTP-1588". After installing and initiating the ptp4l for software time stamping I am receiving the following error. Can anyone please tell me how to resolve this problem

[image]

Thanks

dsb6063, 2019-08-15:
In one of your config files. You are binding to port 53. Change it to 5353. 

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

Ashik19, 2019-08-15:
Can you please tell me where I can get the config file ?

sudo vim /lib/systemd/system/ptp4l.service running this command leading to the following window and my ouster network is eth0

[image]

dsb6063, 2019-08-15:
If you used dnsmasq, I installed here. /etc/ 

sudo nano etc/dnsmasq.conf

check and verify port=5353 is not commented out.

Also, verify dnsmasq is working correctly. You should be able to connect the sensor directly and assign ip address automatically. 

Try this first. 
journalctl -fu dnsmasq

These will restart your services. Or reboot.

sudo systemctl daemon-reload
sudo systemctl restart ptp4l
sudo systemctl status ptp4l

sudo systemctl restart dnsmasq
sudo systemctl daemon-reload
sudo systemctl restart phc2sys
sudo systemctl status phc2sys

sudo systemctl start phc2shm
sudo systemctl status phc2shm

chronyc tracking

journalctl -fu dnsmasq

sudo ptp4l -i eno1 -m -H

Ashik19, 2019-08-15:
Hi @dsb6063 ,

I am using dnsmasq. It's working fine and my sensor is also getting connected using the dnsmasq.
port is 5353
[image]
 phc2shm
[image]

phc2sys
[image]

status ptp4l
[image]

"unknown option production description at line 83 in the eth0 section"

Can you please find any error in the process?

Thank you very much for your valuable time

dsb6063, 2019-08-15:
I believe without knowing more about your setup that you are using one network card. remove the reference to the interface eth0 from line 83. Save and reboot.

Ashik19, 2019-08-16:
Hi @dsb6063 ,

I have solved it we have to add -i (enterface name) in my case eth0 in the "sudo nano /etc/systemd/system/ptp4l.service.d/override.conf"

[image]

But I don't understand when I am running the chronyc tracking it's saying. I have followed all the steps mentioned in the manual.

'506 can not talk to daemon'

Ashik19, 2019-08-16:
HI @dsb6063 ,
I have followed all the steps through the manual.Unfortunately when I am running the chronyc tracking it's saying '506 can not talk to daemon'
[image]

 status ptp4l
[image]

status phc2sys
[image]

status phc2shm
[image]

And when stamping the software getting the following message
[image]

It would be very helpful if you couldguide me here

Thanks

closed 2019-08-19 by Ashik19

mrgransky, 2020-01-17:
For future reference, there's a `Search your computer` icon on the top left of your screen in `Ubuntu 16.04` and `Screenshot` application to avoid taking photo of your screen.

yulan0215, 2021-06-03:
> HI, I have set my lidar timestamp mode to "Time_From_PTP-1588". After installing and initiating the ptp4l for software time stamping I am receiving the following error. Can anyone please tell me how to resolve this problem
> 
> [image]
> 
> Thanks

Hi, do you slove the problem of using PTP based on the software time stamping? I used Jetson Nano which could not support Hardware timestamping.