ouster-lidar/ouster-sdk #100: Extracting Packet Data from rosbag

Ashik19, 2019-08-19:
Hi, 

I am saving my packet data in the ros bag format as ros topic(/os1_node/lidar_packets). How I can unblock the packet data from the rosbag and make it readable? And how we can get the range,ambient and noise data out of the lidar_packet?

Any help is very appreciated.

Thanks

mrgransky, 2020-01-21:
you may have a look at [here](https://github.com/ouster-lidar/ouster_example/issues/146#issue-551331162)!

dgarbuzov, 2020-03-14:
The relevant code is in [batch_to_iter](https://github.com/ouster-lidar/ouster_example/blob/1cf6faf3b993caab483b7e9b25a0f287900acc2a/ouster_client/include/ouster/os1_util.h#L83)

See [os1_cloud_node](https://github.com/ouster-lidar/ouster_example/blob/1cf6faf3b993caab483b7e9b25a0f287900acc2a/ouster_ros/src/os1_cloud_node.cpp#L59) for an example of how it's used.

closed 2020-03-14 by dgarbuzov